erlerobot/gym-gazebo #2: Installation proceedure fails with cuda 8.0

vmayoral, 2016-09-02:
From *Robert Pottorff*:

> just an FYI - the installation proceedure fails with cuda 8.0
> 
> -- Eigen found (include: /usr/include/eigen3)
> CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
>   Could NOT find CUDA: Found unsuitable version "8.0", but required is exact
>   version "7.0" (found /usr/local/cuda)
> Call Stack (most recent call first):
>   /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:313 (_FPHSA_FAILURE_MESSAGE)
>   /usr/share/cmake-2.8/Modules/FindCUDA.cmake:806 (find_package_handle_standard_args)
>   /usr/local/share/OpenCV/OpenCVConfig.cmake:48 (find_package)
>   /usr/local/share/OpenCV/OpenCVConfig.cmake:276 (find_host_package)
>   ar_track_alvar/CMakeLists.txt:28 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/mnt/pccfs/downloads/gym/gym/envs/gazebo/installation/catkin_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/mnt/pccfs/downloads/gym/gym/envs/gazebo/installation/catkin_ws/build/CMakeFiles/CMakeError.log".
> Invoking "cmake" failed
> ## ROS workspace compiled ##

Nestor94Gonzalez, 2016-09-05:
I suspect Robert is not using Ubuntu 14.04. However, I would try going back to 7.5 (latest official version and currently working in our machines). Another solution could be deleting 'libopencv-dev' from the setup [script](https://github.com/erlerobot/gym-gazebo/blob/master/gym/envs/gazebo/installation/setup.bash) and installing a cuda 8 compatible version. 

Step by step installation [instructions](https://github.com/erlerobot/gym-gazebo/blob/master/gym/envs/gazebo/INSTALL.md#step-by-step-installation) may help.

Nestor94Gonzalez, 2016-09-14:
Update, cuda 8.0 tested and working.

closed 2017-11-20 by vmayoral