erlerobot/gym-gazebo #3: Error while running circuit2_turtlebot_lidar_qlearn

karanchawla, 2016-09-05:
Trying to run the first example for the  turtlebot autonomous navigation. Screen shot of the terminal is attached. 
[image]

Also I plan to implement the same for an AR Drone 2.0 probably by using tum_simulator along with ar_drone autonomy, should I just follow what you guys did with the Erle Copter? 

Thanks.

IkerZamora, 2016-09-05:
Run
```
sudo python setup.py build install
````
in the root directory of the repository.

IkerZamora, 2016-09-05:
About the AR Drone 2.0 simulation... sure! Feel free to check out the Erle-Copter environment! Although I wouldn't suggest using the current version as an example. Since it's still Work In Progress and not very stable. But we are currently working on it, so stay tuned! :smile: 

Also, take into account that the simulation architecture for the Erle-Copter is pretty special; and a bit different from how I guess tum_simulator works. It needs to be run alongside with ArduPilot, which controls the copter using a step lock mechanism (more info about it [here](http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html)). 

But it sure will give you an idea of where to start your implementation. Feel free to ask for advice otherwise. And also feel free to contribute to the community and submit your work! :grin:

karanchawla, 2016-09-06:
Hey @IkerZamora thanks for the help. Kinda foolish I didn't build the gym library first. Anyway so I'm done with setting up both and gazebo environments. When I try to run 

`python circuit2_turtlebot_lidar_qlearn.py`

I get the following error: 
`IOError: File /usr/local/lib/python2.7/dist-packages/gym-0.2.3-py2.7.egg/gym/envs/gazebo/assets/launch/GazeboCircuit2TurtlebotLidar_v0.launch does not exist`
From my understanding it's unable to locate the Gazebo launch file in the assets folder, which has something to do with the env variables not set up properly. Do you know how I can fix this?

karanchawla, 2016-09-06:
Also thanks for the advice on the AR Drone simulation. Yes, I've to catch up on lot of reading regarding the idea. But I think it should be possible to do it with a lot of coffee and sleepless night :P 

And yes, once I have all this set up correctly I'll go back to docs read up on stuff, come up with a systems level plan and maybe discuss that with you. 

Thanks.

Cheers,
Karan

IkerZamora, 2016-09-06:
>I get the following error: 
`IOError: File /usr/local/lib/python2.7/dist-packages/gym-0.2.3-py2.7.egg/gym/envs/gazebo/assets/launch/GazeboCircuit2TurtlebotLidar_v0.launch does not exist`
From my understanding it's unable to locate the Gazebo launch file in the assets folder, which has something to do with the env variables not set up properly. Do you know how I can fix this?

That shouldn't be env variables issue, that launch file should be there. But just to make sure, load these:

```
cd gym/envs/gazebo/installation
bash turtlebot_setup.bash
source catkin_ws/devel/setup.bash
```
Anyway, what does your installation directory look like? Could you please share the tree structure?

```
tree /usr/local/lib/python2.7/dist-packages/gym-0.2.3-py2.7.egg/gym/envs/gazebo
```

karanchawla, 2016-09-06:
/usr/local/lib/python2.7/dist-packages/gym-0.2.3-py2.7.egg/gym/envs/gazebo
├── gazebo_circuit2_turtlebot_lidar.py
├── gazebo_circuit2_turtlebot_lidar.pyc
├── gazebo_circuit_turtlebot_lidar.py
├── gazebo_circuit_turtlebot_lidar.pyc
├── gazebo_env.py
├── gazebo_env.pyc
├── gazebo_maze_erlerover_lidar.py
├── gazebo_maze_erlerover_lidar.pyc
├── gazebo_maze_turtlebot_lidar.py
├── gazebo_maze_turtlebot_lidar.pyc
├── gazebo_office_erlecopter_lidar.py
├── gazebo_office_erlecopter_lidar.pyc
├── gazebo_round_turtlebot_lidar.py
├── gazebo_round_turtlebot_lidar.pyc
├── __init__.py
└── __init__.pyc

There you go, and yes I loaded those before attempting to run the python file.

IkerZamora, 2016-09-06:
Okay, I know what's happening. It's an installation issue, `assets/` and `worlds/` folders are missing in the `setup.py`. I'm out now, can you fix it @Nestor94Gonzalez ?

karanchawla, 2016-09-06:
We'll just edit the setup.py from-
`package_data={'gym': ['envs/mujoco/assets/*.xml', 'envs/classic_control/assets/*.png', 'envs/doom/assets/*.cfg']},` to 
`package_data={'gym': ['envs/mujoco/assets/*.xml', 'envs/classic_control/assets/*.png', 'envs/doom/assets/*.cfg','envs/gazebo/assets/launch/*.launch','envs/gazebo/assets/worlds/*.launch']},`
right ? And run setup.py again?

Nestor94Gonzalez, 2016-09-06:
Hi @karanchawla , 
Yes, replace that line with [this](https://gist.github.com/Nestor94Gonzalez/678b3754b691ebb6e92b3b4ea313d689) one. We updated our development repository but not this one sorry.

karanchawla, 2016-09-06:
Thanks @Nester94Gonzalez 
I also checked my code, works just fine. :)

closed 2016-09-07 by karanchawla

ozdenurucar, 2018-11-23:
Hi, I've been trying to 'circuit2_turtlebot_lidar_qlearn' but I'm constantly getting the following error. I'm using the gazebo 7.6 which comes with ros-fulldesktop. I installed Gazebo8 but the gazebo8.6 version was installed. 8.1.1 was required for gym applications. Could I get this error because of the gazebo version?

[image]

nzlz, 2018-11-24:
I don't think this is a gazebo version related issue. Did you install gym-gazebo?
`cd gym-gazebo /
sudo pip install -e .`  
Please follow all the installation steps available [here](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md).

ozdenurucar, 2018-11-24:
I've run all the steps, but when I run it again, I get an error like this. What is the reason?
[image]

ozdenurucar, 2018-11-24:
I've done all the steps again, but I'm getting an error like this right now. Can you help me? 

[image]

nzlz, 2018-11-25:
`GYM_GAZEBO_WORLD_CIRCUIT2` variable is set when you run the [turtlebot_setup.bash](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/installation/turtlebot_setup.bash) file. You can se [here ](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#run-example-for-qlearn) it is a requirement that you load those variables before you launch the learning script.

ozdenurucar, 2018-11-26:
Thank you  so much. İt worked, but when I try again, I get an error like the following.

[image]

nzlz, 2018-11-26:
Use python2, not 3. Also know that the path to the turtlebot package (not turtlebot3), should be located in `/home/XYZ/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot`. Please follow the [instructions](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md) carefully.

ozdenurucar, 2018-11-26:
I'm using this code `python circuit2_turtlebot_lidar_qlearn.py ` also not python3.  I checked all the steps as you said, fix the file path and run the `bash turtlebot_setup.bash` command again. It's successfully. 
And I try again, but not worked :( I get an error the following

[image]

~/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/kobuki$ tree
.
├── kobuki
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   └── package.xml
├── kobuki_auto_docking
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   ├── include
│   │   └── kobuki_auto_docking
│   │       └── auto_docking_ros.hpp
│   ├── launch
│   │   ├── activate.launch
│   │   ├── auto_dock_with_safe_keyop.launch
│   │   ├── compact.launch
│   │   ├── minimal.launch
│   │   └── standalone.launch
│   ├── package.xml
│   ├── param
│   │   ├── auto_docking.yaml
│   │   └── cmd_vel_mux.yaml
│   ├── plugins
│   │   └── nodelet_plugins.xml
│   ├── scripts
│   │   ├── DockDriveActionClient.py
│   │   ├── DockDriveControl.py
│   │   └── getDockIR.py
│   └── src
│       ├── auto_docking_ros.cpp
│       └── nodelet.cpp
├── kobuki_bumper2pc
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   ├── include
│   │   └── kobuki_bumper2pc
│   │       └── kobuki_bumper2pc.hpp
│   ├── launch
│   │   └── standalone.launch
│   ├── package.xml
│   ├── plugins
│   │   └── nodelet_plugins.xml
│   └── src
│       └── kobuki_bumper2pc.cpp

thanks for all your help.

nzlz, 2018-11-26:
Take a look at https://github.com/erlerobot/gym-gazebo/issues/108, where Mostafa references https://github.com/erlerobot/gym-gazebo/issues/110. Also You can check [this](https://answers.ros.org/question/279219/turtlebot-not-moving/?answer=279349#post-id-279349) but I really would not recommend that solution. 

Also make sure you have the latest version of the repository and you could also try comparing the installation steps you followed with the ones for Ubuntu 18.04, which I have tested recently. 

Make sure every time the main script crashes you end gazebo related processes. This is the alias I use: `echo "alias killgazebogym='killall -9 rosout roslaunch rosmaster gzserver nodelet robot_state_publisher gzclient'" >> ~/.bashrc
`

nzlz, 2018-12-04:
Hi @ozdenurucar ,

The error is a known bug, but it only means the Turtlebot failed to spawn the first time. After this, you can see the Turtlebot is moving correctly and manages to spawn, move, learn... So everything is working as it is supposed to.

Gazebo is working, in fact, the gzserver is running. In order to get faster simulation speeds the gzclient (gazebo GUI) is not launched by default. You need to run the `gzclient` in a new terminal if you want to see the robot moving. 

If you want to launch the GUI by default, please check https://github.com/erlerobot/gym-gazebo/issues/188#issuecomment-443544491 .

ozdenurucar, 2018-12-04:
Thank you so much! It's working.

ozdenurucar, 2018-12-30:
Hi, I installed Cartographer for SLAM  applications then gym-gazebo
 didn't work. I uninstall all cartographer's  package and gym-gazebo packages then I try again but get the following error while running this codes 
```
cd gym-gazebo/gym_gazebo/envs/installation
bash setup_melodic.bash
```

nzlz, 2018-12-31:
Fastest way, [remove hector_gazebo](https://github.com/erlerobot/gym-gazebo/blob/7c63c16532f0d8b9acf73663ba7a53f248021453/gym_gazebo/envs/installation/gazebo_ros_melodic.repos#L38) package or ignore its compilation. You don't need it if you are not using the ErleRover.

ozdenurucar, 2018-12-31:
It's compiled but I get some error again 

[image]

nzlz, 2018-12-31:
I don't know what is your exact issue or what did u install. But if you just want to execute gym-gazebo, make sure you don't have unnecessary sources in your `~/.bashrc`. Also, why don't you upgrade `protobuf` as the error says?