erlerobot/gym-gazebo #5: Turtlebot not learning using CNN

Nestor94Gonzalez, 2016-09-13:
# Turtlebot not learning using CNN

_https://github.com/erlerobot/gym-gazebo/_

We are trying to teach the Turtlebot robot how to avoid obstacles  in [this](https://cloud.githubusercontent.com/assets/16973377/18428362/d7b08c76-78cd-11e6-989c-f24850ffefd3.jpg) circuit using solely the camera as input to a convolutional neural net. We have tried many network configurations but we have not achieved good results so far. Installation instructions are available [here](https://github.com/erlerobot/gym-gazebo#installation). We are using [Keras+Theano](https://github.com/erlerobot/gym-gazebo#keras-and-theano-installation).

Summary:
- We provide the CNN a 32x32 greyscale image (watch video below). 
- Code: [script](https://github.com/erlerobot/gym-gazebo/blob/master/examples/scripts_turtlebot/circuit2c_turtlebot_camera_dqn.py), [environment](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_circuit2c_turtlebot_camera_nn.py)
- 3 [Actions](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_circuit2c_turtlebot_camera_nn.py#L74-L88): Left, Right and Forward.
- [Rewards](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_circuit2c_turtlebot_camera_nn.py#L147-L157).
- Network structure: [link](https://github.com/erlerobot/gym-gazebo/blob/master/examples/scripts_turtlebot/circuit2c_turtlebot_camera_dqn.py#L88-L111)

Things we tried but did not show improvement:
- Stack last 4 greyscale images, generating a [4x32x32 input](https://gist.github.com/IkerZamora/9531d52393a963589e4bfa44d452b37e).
- Different network configurations.
- Different wall colours (left-right walls and the floor had different colours).
- Different camera angles.
- Different reward system.
- 21 actions instead of 3.
- Start with lower exploration rate.
- Spawn the robot in different positions.
- Many others.

Note: In some trainings the robot takes the first turn to the right, this could be due to [this](http://answers.ros.org/question/243549/turtlebot-spinning-in-place-not-going-straight-when-speeding-up-simulation/) issue.

---
Repository : https://github.com/erlerobot/gym-gazebo
Video ep 0 : https://youtu.be/uOwfdUCXsbw
Video ep 1000 : https://youtu.be/Uv8aGFybKu4
Episode 1000 plot : 
[image]
Here you can see another example, where we feed the CNN with a stack of the last 4 images. In the best case scenario, the robot makes successfully the first turn of the [circuit](https://cloud.githubusercontent.com/assets/7148841/18203702/dbae30f6-7118-11e6-8011-6c85b31de4e6.png). After the 3000th episode the learning gets stuck. It's also worth to mention that we give 1 point reward when turning, 5 when going straight and -500 when hitting a wall.
[image]
The following is another example. Here changed the camera angle too.
[Youtube video](https://youtu.be/E8v8n4x-Up0)
[Network structure](https://gist.github.com/Nestor94Gonzalez/7dfa85d0dbe8d7ac53674c9e28d4456b)

tlbtlbtlb, 2016-09-13:
The code looks great!

You may just need to crank up the memory and wait longer. Vlad Mnih's Atari work (http://arxiv.org/pdf/1312.5602v1.pdf)  "trained for a total of 10 million frames and used a replay
memory of one million most recent frames". I'd guess your environment is within a couple orders of magnitude of difficulty of the maze-type Atari games.

Also, your finalExplorationRate=0.1 might be too high. Should it able to navigate the maze reliably while making 10% random moves?

Better yet, your environment may need long-range exploration. A good approach is http://arxiv.org/abs/1602.04621.

Nestor94Gonzalez, 2016-09-23:
We found and solved a bug in the code (the loss value was not being properly calculated) and now the robot is learning. We will update the code soon. Thanks anyway @tlbtlbtlb !

closed 2016-09-23 by Nestor94Gonzalez

patwnz, 2017-08-14:
I tried out the code, however I was not able to successfully train the robot using solely the camera. Where did you changed the bug regarding the loss value? How long does it take to get decent results? Thanks in advance.