erlerobot/gym-gazebo #10: Docker | Gazebo gzclient GUI not displaying on OS X El Capitan

ceobillionaire, 2016-10-02:
On _OS X El Capitan_ (version 10.11.6), when running:

```
export GAZEBO_MASTER_IP=$(sudo docker inspect --format '{{ .NetworkSettings.IPAddress }}' "id of running container")
export GAZEBO_MASTER_URI=$GAZEBO_MASTER_IP:11345
gzclient
```
Nothing happens (I have Gazebo version 7.3.1 installed locally).

Launching `gazebo` alone is working.

vmayoral, 2016-10-02:
+1, reproduced it and got the same result. No response for `gzclient`.

Nestor94Gonzalez, 2016-10-02:
This feature has been tested just in Ubuntu14.

rdaries29, 2017-08-18:
Hello,

I have encountered the same error on OS X Sierra (version 10.12.3). When I launch the _circuit2_turtlebot_lidar_qlearn.py_ file, it does perform episodes but I am unable to visualise the environment. Launching gazebo alone is working. The command gzclient is unresponsive for a second.

[image]

AliBaheri, 2017-10-20:
Same problem here on the same system!

vmayoral, 2017-10-23:
Docker container is not being maintained anymore. Refer to https://github.com/erlerobot/gym-gazebo/issues/44 for more details.

closed 2017-10-23 by vmayoral