erlerobot/gym-gazebo #11: scripts_erlecopter

ceobillionaire, 2016-10-04:
In Docker on Ubuntu 14.04, when I try to run: `python erlecopter_hover_qlearn.py`

I get:

> Traceback (most recent call last):
>   File "erlecopter_hover_qlearn.py", line 7, in <module>
>     import pandas
> ImportError: No module named pandas

Steps to reproduce:

```
sudo docker run -it erlerobotics/gym-gazebo
cd /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation
bash erlecopter_setup.bash
cd /usr/local/gym/gym-gazebo/examples/scripts_erlecopter
python erlecopter_hover_qlearn.py
```

Nestor94Gonzalez, 2016-10-04:
Install pandas : `sudo pip install pandas`

ceobillionaire, 2016-10-04:
Thanks! :+1: 

Then, I get:

```
[gazebo_gui-4] process has died [pid 257, exit code 1, cmd /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzclient __name:=gazebo_gui __log:=/root/.ros/log/212e5dfa-8a45-11e6-ae81-0242ac110002/gazebo_gui-4.log].
log file: /root/.ros/log/212e5dfa-8a45-11e6-ae81-0242ac110002/gazebo_gui-4*.log
libdc1394 error: Failed to initialize libdc1394
```

Nestor94Gonzalez, 2016-10-04:
Try using a fake screen: `xvfb-run -s "-screen 0 1400x900x24" bash` . libdc1394 error will still appear but the simulation should work. 

Also note that the docker image is useful for training in any OS, if you want to develop the full installation in ubuntu 14 is recommended.

ceobillionaire, 2016-10-04:
Putting all together, I run:

```
sudo docker run -it erlerobotics/gym-gazebo
sudo pip install pandas
cd /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation
xvfb-run -s "-screen 0 1400x900x24" bash
bash erlecopter_setup.bash
source catkin_ws/devel/setup.bash
cd /usr/local/gym/gym-gazebo/examples/scripts_erlecopter
python erlecopter_hover_qlearn.py
```

Which gives (only) lots of:

```
( . . . )
[ INFO] [1475603940.001280182, 421.842500000]: udp0: Remote address: 127.0.0.1:37859
[ INFO] [1475603940.002060423, 421.842500000]: udp0: Remote address: 127.0.0.1:53007
[ INFO] [1475603940.002380702, 421.842500000]: udp0: Remote address: 127.0.0.1:44046
[ INFO] [1475603940.581925296, 422.832500000]: udp0: Remote address: 127.0.0.1:37859
[ INFO] [1475603940.583379558, 422.845000000]: udp0: Remote address: 127.0.0.1:53007
[ INFO] [1475603940.584147355, 422.845000000]: udp0: Remote address: 127.0.0.1:44046
[ INFO] [1475603941.116182397, 423.835000000]: udp0: Remote address: 127.0.0.1:37859
[ INFO] [1475603941.118835788, 423.845000000]: udp0: Remote address: 127.0.0.1:53007
[ INFO] [1475603941.121412218, 423.845000000]: udp0: Remote address: 127.0.0.1:44046
[ INFO] [1475603941.922049472, 424.842500000]: udp0: Remote address: 127.0.0.1:37859
[ INFO] [1475603941.922611956, 424.842500000]: udp0: Remote address: 127.0.0.1:53007
[ INFO] [1475603941.923312242, 424.842500000]: udp0: Remote address: 127.0.0.1:44046
( . . . )
```

Is there a way to get useful info such as the "observation" and "reward"?

Nestor94Gonzalez, 2016-10-04:
That information is printed every episode. You can try the turtlebot examples, which are the ones I tested using docker. @IkerZamora please check this erle-copter issue.

Abhi001vj, 2017-06-21:
Can we see the simulations on Gazebo.

Kjell-K, 2017-08-24:
@Nestor94Gonzalez I get the same output as @ceobillionaire
After:

    env = gym.make('GazeboCircuitTurtlebotLidar-v0')
    outdir = '/tmp/gazebo_gym_experiments'
    env = gym.wrappers.Monitor(env, directory=outdir, force=True, write_upon_reset=True)
    
    print "I will not be printed"

the script is not progressing further. Any solutions?

closed 2017-11-20 by vmayoral