erlerobot/gym-gazebo #12: Error when compiling ar_track_alvar in Ubuntu 16.04

peterzcc, 2016-10-16:
I finally successfully installed the repository in Ubuntu 16.04. It's really an impressive work and I really appreciate it.
One thing I encountered was that there were a lot of error about undeclared function calls related to opencv. I finally side-stepped them by removing all the opencv related function calls. May I ask if this could cause any problem?

Nestor94Gonzalez, 2016-10-16:
Scripts using the camera won't work since they use opencv (import cv2 in python). I would try making a proper installation of opencv for 16.04 or using other image processing tools. However I can't tell if the gym would work with that change since we haven't used that Ubuntu version , you can test simple scripts like Q-learn and Sarsa first.

peterzcc, 2016-10-17:
@Nestor94Gonzalez Thanks a lot for your reply!

I have OpenCV 3.1.0 installed and therefore the gym works well, and are only part of serveral OpenCV function calls that are not compatible in the files Camera.cpp, MultiMarkerBundle.cpp, and MultiMarkerBundle.cpp. I will try replacing the files with [this repository](https://github.com/mariosbikos/alvar_updated/tree/master/src) to see it it gets better.

Faur, 2017-05-23:
@peterzcc  were you successful?

closed 2017-11-20 by vmayoral