erlerobot/gym-gazebo #20: Wrapper monitor error solution

forthtemple, 2017-04-09:
If anyone is having an issue with monitoring, monitoring in the latest version of gym been replaced by wrappers, therefore monitoring will not work with the latest gym. You can fix it in the file:

circuit2_turtlebot_lidar_qlearn.py

by replacing the line

env.monitor.start(outdir, force=True, seed=None)
with
 env = gym.wrappers.Monitor(env, directory=outdir, force=True, write_upon_reset=True)

and comment out the line 

env.monitor.flush(force=True).

In the file display_plot.py replace the line 

results = gym.monitoring.monitor.load_results(self.outdir)
with
results = monitoring.load_results(self.outdir)

and display_plot.py should work.

allanis79, 2017-05-26:
I have made the above changes to my code and when I run the 

> circuit2_turtlebot_lidar_qlearn.py

I am getting the follwoing error: 

`Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 89, in <module>
    observation = env.reset()
  File "/usr/local/lib/python2.7/dist-packages/gym/core.py", line 107, in reset
    return self._reset()
  File "/usr/local/lib/python2.7/dist-packages/gym/wrappers/monitoring.py", line 39, in _reset
    self._before_reset()
  File "/usr/local/lib/python2.7/dist-packages/gym/wrappers/monitoring.py", line 190, in _before_reset
    self.stats_recorder.before_reset()
  File "/usr/local/lib/python2.7/dist-packages/gym/monitoring/stats_recorder.py", line 68, in before_reset
    raise error.Error("Tried to reset environment which is not done. While the monitor is active for {}, you cannot call reset() unless the episode is over.".format(self.env_id))
gym.error.Error: Tried to reset environment which is not done. While the monitor is active for GazeboCircuit2TurtlebotLidar-v0, you cannot call reset() unless the episode is over.`

So i added the line 

`env. close` in place of `env.monitor.flush(force=True)`

If i add the above it doesn't throw any error but it looks like the robot is not learning anything and I don't see anything on the graph. I kept it running for 400 episodes and the robot just roatates in the same place. 

can you help me? @forthtemple

su-si, 2017-05-28:
I didn't get the same error as @allanis79, but the monitor did not log anything into the "outdir".
It seemes to be solved now after adding a
`env._flush(force=True)`
in place of `env.monitor.flush(force=True)`

I think if "flush" is removed, it only logs at the end. 

@allanis79 , I'm new to this as well, but maybe check whether you changed anything else? Line 89 for the line "observation = env.reset()" seems really high.

YijiongLin, 2017-06-06:
Actually when We In the file display_plot.py replace the line
results = gym.monitoring.monitor.load_results(self.outdir)
with
results = monitoring.load_results(self.outdir). It still doesn't work.

But when we replace the line
results = gym.monitoring.monitor.load_results(self.outdir)
with
results =gym.monitoring.load_results(self.outdir)
 It works.

We thought there were lots of changes in the gym module.

closed 2017-11-20 by vmayoral