erlerobot/gym-gazebo #22: fail to run the turtlebot example

onlytailei, 2017-04-10:
Firstly I start 
```
cd gym_gazebo/envs/installation
bash turtlebot_setup.bash
```

Then 
```
cd examples/scripts_turtlebot
python circuit2_turtlebot_lidar_qlearn.py
```
There is an error
```
[ERROR] [1491817752.050551945, 124.140000000]: Failed to load nodelet [/cmd_vel_mux] of type [yocs_cmd_vel_mux/CmdVelMuxNodelet] even after refreshing the cache: Could not find library corresponding to plugin yocs_cmd_vel_mux/CmdVelMuxNodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1491817752.050594806, 124.140000000]: The error before refreshing the cache was: Could not find library corresponding to plugin yocs_cmd_vel_mux/CmdVelMuxNodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1491817752.050726655, 124.140000000]: Failed to load nodelet '/cmd_vel_mux` of type `yocs_cmd_vel_mux/CmdVelMuxNodelet` to manager `mobile_base_nodelet_manager'
[cmd_vel_mux-5] process has died [pid 8445, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load yocs_cmd_vel_mux/CmdVelMuxNodelet mobile_base_nodelet_manager cmd_vel_mux/output:=mobile_base/commands/velocity __name:=cmd_vel_mux __log:=/home/tai/.ros/log/f0cc8c0a-1dd2-11e7-a470-5404a6efcc23/cmd_vel_mux-5.log].
log file: /home/tai/.ros/log/f0cc8c0a-1dd2-11e7-a470-5404a6efcc23/cmd_vel_mux-5*.log

```

From **gzclient**, I can see the turtlebot in the gazebo world, but it is not moving.

Then I start 
```
python display_plot.py
```
There is another error:
```
$ python display_plot.py 
Traceback (most recent call last):
  File "display_plot.py", line 106, in <module>
    plotter.plot(full=True)
  File "display_plot.py", line 37, in plot
    results = gym.monitoring.monitor.load_results(self.outdir)
AttributeError: 'module' object has no attribute 'monitor'

```

forthtemple, 2017-04-14:
Maybe look at my solution at

https://github.com/erlerobot/gym-gazebo/issues/20

Gym has changed the way monitors work and has broken display_plot.py

ElizabethDuo, 2017-08-17:
Also failed run turtlebot examples. I did the same parts as onlytailei did and showed an error like this
root@elizabeth:~/gym-gazebo/examples/scripts_turtlebot# python circuit2_turtlebot_lidar_qlearn.py
[2017-08-17 13:10:56,872] Making new env: GazeboCircuit2TurtlebotLidar-v0
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/root/gym/gym/envs/registration.py", line 161, in make
    return registry.make(id)
  File "/root/gym/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/root/gym/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/root/gym/gym/envs/registration.py", line 17, in load
    result = entry_point.load(False)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/root/gym-gazebo/gym_gazebo/envs/__init__.py", line 8, in <module>
    from gym_gazebo.envs.gazebo_circuit2c_turtlebot_camera_nn import GazeboCircuit2cTurtlebotCameraNnEnv
  File "/root/gym-gazebo/gym_gazebo/envs/gazebo_circuit2c_turtlebot_camera_nn.py", line 20, in <module>
    import skimage as skimage
ImportError: No module named skimage
I wonder is there something wrong with my installation or with my ros.
And I try to <ctrl+b>gzclient<ctrl+b>, and there is another error like this
Msg Waiting for master..............................
Error [ConnectionManager.cc:116] Failed to connect to master in 30 seconds.
Error [gazebo.cc:59] Unable to initialize transport.
Error [gazebo.cc:203] Unable to setup Gazebo
But I have already roscore to start ros.
I will be very appreciate if you help me with this problem!