erlerobot/gym-gazebo #24: QUESTION: How is turtlebot speedup acheived?

schmittlema, 2017-04-15:
I was wondering how you were able to run turtlebot faster than normal speed. It looks like it may have to do with pausing physics and then unpausing, but I wasn't 100% sure. Thanks!

TolgaOk, 2017-04-22:
There is a "real time update rate" parameter in the gazebo GUI under "Physics". You can set it to 0 to make it as fast as possible. Its default value is 1000.
You can also use the service "/gazebo/set_physics_properties" with arguments; time_step: 0.001 and max_update_rate: 0.0 (which sets it to max possible).

schmittlema, 2017-05-16:
Ok sweet thank you! For the q learning with the speed up on the simulator how do you handle the learning running at a different speed then simulator? Like if the simulator is set to 1000 or 0 isnt the q learning still running at the same rate for both?

TolgaOk, 2017-05-16:
Exactly, since its pausing and unpausing, the time between them will be decreased. Depending on your hardware, and deep learning architecture, total process will speed up but not as high as simulator.

schmittlema, 2017-05-16:
Awesome thanks!

closed 2017-05-16 by schmittlema