erlerobot/gym-gazebo #25: Files related to 2d Occupancy Grid (.pgm, .yaml)

surfertas, 2017-04-28:
Hi, thanks for this. I am assuming the .world files were created using Gazebo. I was wondering if your team needed the 2D occupancy grid map representation that RViz would use for a localization algorithm to process. 

If this was the case, do you have any suggestions on efficiently generating the 2D occupancy grid from the given Gazebo .world file, as opposed to using gmapping and manually teleoping around in RVIZ hoping for a decent map? 

Thanks a lot.

Nestor94Gonzalez, 2017-04-28:
Hi @surfertas , I don't see why gmapping is a bad idea here, once the robot learns to avoid obstacles there will be no need of manually teleoperating it. 
About getting the grid from the world file.. I think it is a better idea to create it from sensory information, but I may have misunderstood you.
Thanks for the support.

surfertas, 2017-04-29:
@Nestor94Gonzalez Thanks very much for the quick response. Gmapping is great. The only thing is that gmapping requires manually generating the map via sensory input obtained by teleoping the turtlebot around the environment.

My intention is create many maps (with obstacles placed randomly) in Gazebo and generate 2D occupancy maps (.pgm, .yaml files) that I can pass to RVIZ for navigation tasks. Having to use gmapping and teleoping to create the maps from sensory inputs would not be practical.

The reason I am attempting this is that I would like to generate data related to navigation in simulation that I can apply a supervision learning model.

I thought you guys may have had a similar need in generating 2D occupancy grids thus may have already worked on a solution.

vmayoral, 2017-11-20:
Closing since the conversation has been inactive for a while. Feel free to re-open it if necessary.

closed 2017-11-20 by vmayoral