erlerobot/gym-gazebo #28: ImportError: No module named msg

allanis79, 2017-05-23:
Hello,

I followed the documentation to install the gym-gazebo. buw when i run the " circuit2_turtlebot_lidar_qlearn.py" , I am gettign the follwoing error:

`[2017-05-23 11:14:34,399] Making new env: GazeboCircuit2TurtlebotLidar-v0
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 161, in make
    return registry.make(id)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 17, in load
    result = entry_point.load(False)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/home/allani/Desktop/gym-gazebo/gym_gazebo/envs/__init__.py", line 12, in <module>
    from gym_gazebo.envs.gazebo_erlecopter_hover import GazeboErleCopterHoverEnv
  File "/home/allani/Desktop/gym-gazebo/gym_gazebo/envs/gazebo_erlecopter_hover.py", line 14, in <module>
    from mavros_msgs.msg import OverrideRCIn
ImportError: No module named msg`

Can anyone help me how to solve this?

Thanks

bhairavmehta95, 2017-07-17:
Have you made sure to build the catkin_ws? I had a similar issue in that I had to build only the mavros_msgs, source the workspace, and then build only the driver_base package before it compiled.

DanielLSM, 2017-07-19:
Hello,
I have the same problem, tried to build the catkin_ws and got  

Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Any clue?
Thanks in advance

bhairavmehta95, 2017-07-19:
@allanis79 

What worked for me was:

```
catkin_make --pkg mav_msgs
source devel/setup.bash
catkin_make --pkg driver_base
source devel/setup.bash
catkin_make -j8
source devel/setup.bash
```

@DanielLSM Can you post all of your stack, specifically the messages labelled ERROR when you build with catkin_make?

closed 2017-11-20 by vmayoral