erlerobot/gym-gazebo #29: Issues with Anaconda

Faur, 2017-06-02:
When following the installation instructions on https://github.com/erlerobot/gym-gazebo#gazebo I have problems that I think are due to the fact that I am using Anaconda (2.7).

When I run `gym_gazebo/envs/installation/setup.bash` I see several references to the original version of python, despite adding anaconda in my `.bashrc` file.

Am I doing something wrong, or how do I use gym-gazebo with Anaconda?

bhairavmehta95, 2017-07-28:
Can you post your `.bashrc` in here?

When you run setup.bash, it's unclear from your post where pip is actually installing the packages.

vmayoral, 2017-11-20:
Please use Python 3 for newer installations. Getting it working with `conda` should be doable however the amount of work you'll need to put into getting PATHs appropriately is unknown. We haven't explored it.

closed 2017-11-20 by vmayoral