erlerobot/gym-gazebo #30: Ubuntu Build Requirements

bhairavmehta95, 2017-07-12:
I was not able to get this working with a fresh new 14.04 install using the current Ubuntu / Gazebo install instructions.

Has anyone else had this problem? I would love to share my solution, but not if it only did not work because I may have skipped a step or done something wrong.

silverjoda, 2017-09-18:
No I haven't had success either. The catkin_make crashes at not finding some nintendo wee driver and I've tried installing it but that doesn't work either.

bhairavmehta95, 2017-09-18:
I ended up getting it to work. Have you installed `ros-indigo-joystick-drivers` with `sudo apt-get`?

I did this with the binary rather than in the catkin_ws and it worked for me.

closed 2017-09-18 by bhairavmehta95

silverjoda, 2017-09-18:
Yes I have that latest joystick drivers. What do you mean with binary?