erlerobot/gym-gazebo #31: GazeboCircuit2cTurtlebotCameraNnEnv-v0

patwnz, 2017-08-16:
Hi All,

I'm referring to the closed issue: https://github.com/erlerobot/gym-gazebo/issues/5

I tried to train the turtlebot using this environment, but I can't achieve good results. Therefore my question is, did anyone of you successfully trained an agent using this scenario? If so, how did you achieved descent results? 

Thank you in advance!

ElizabethDuo, 2017-08-17:
Hi I want to ask that have you ever meet  the error when 

> python circuit2_turtlebot_lidar_qlearn.py
or run other examples, my error is
```
[2017-08-17 13:06:40,451] Making new env: GazeboCircuit2TurtlebotLidar-v0
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/root/gym/gym/envs/registration.py", line 161, in make
    return registry.make(id)
  File "/root/gym/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/root/gym/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/root/gym/gym/envs/registration.py", line 17, in load
    result = entry_point.load(False)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/root/gym-gazebo/gym_gazebo/envs/__init__.py", line 8, in <module>
    from gym_gazebo.envs.gazebo_circuit2c_turtlebot_camera_nn import GazeboCircuit2cTurtlebotCameraNnEnv
  File "/root/gym-gazebo/gym_gazebo/envs/gazebo_circuit2c_turtlebot_camera_nn.py", line 20, in <module>
    import skimage as skimage
ImportError: No module named skimage
```
Do you know what kind of error is this and how to fix this? 
Thanks a lot!

patwnz, 2017-08-17:
Hi,
from the error code it seems that you don't have the module scikit-image installed. Please install it using the bash using pip: `pip install -U scikit-image`
Hope that helps.

darshisen, 2017-08-20:
I'm getting a different issue running this script -- the Episodes (Epochs) are not iterating and I see the following in the log file:

[INFO] [WallTime: 1503194320.079259] [0.000000] Loading model xml from ros parameter
[INFO] [WallTime: 1503194320.084702] [0.000000] Waiting for service /gazebo/spawn_urdf_model
[INFO] [WallTime: 1503194320.387306] [62.860000] Calling service /gazebo/spawn_urdf_model
[INFO] [WallTime: 1503194320.404192] [83.910000] Spawn status: SpawnModel: Model pushed to spawn queue, but spawn service timed out waiting
 for model to appear in simulation under the name mobile_base

I am running it in the Docker container.  Any help would be much appreciated!

darshisen, 2017-08-21:
I got my model running - needed to run the command:

 xvfb-run -s "-screen 0 1400x900x24" bash

However, after 7000 episodes, the cumulative reward is stuck at around 22.  So, not learning.  Any ideas?

patwnz, 2017-08-22:
No, I still don't have any idea why it's not learning. As I mentioned before, in the closed issue someone mentioned a bug in the loss calculation, but this didn't helped me. Unfortunately, still no reply. However, if you find something out, please share it.

darshisen, 2017-08-22:
The guy who said it was an issue with the loss function seemed to have
already commited his code in github.  I think we have his latest changes.
But still not learning.

What I did do was dump the image frames coming back from the simulator.
And when I looked at them they all look corrupted.  That would explain the
CNN not learning.

Not sure what to do at this point.  Maybe upgrade the camera module?

patwnz, 2017-08-22:
I checked the commit history and couldn't find anything related to the loss bug. Anyway, I think the loss function is properly calculated.

What do you mean by corrupted? What is wrong with your output images? Did you used the camera_visualizer.py in order to check the images?

darshisen, 2017-08-22:
The only commit related to change in loss function around the time he made
that post is this:

https://github.com/erlerobot/gym-gazebo/commit/3cd72b8f9f0ddcb4640ff62f648490bf37e19cb8

I assume this is what he meant.

I'm using gym-gazebo in a docker container and can't get the camera
visualizer to work right - seems like it's not getting messages.

patwnz, 2017-08-23:
He didn't changed the loss function, he just changed the optimizer from Adam to RMSprop. Anyway, I guess the turtlebot is not learning due to the choice of the reward function.

closed 2017-10-25 by pmwenzel