erlerobot/gym-gazebo #35: ResourceNotFound: turtlebot_gazebo when using Gazebo8 under ROS kinetic

jeasinema, 2017-10-11:
Hi,
I've successfully set up the needed environment for `gym-gazebo`, but when I try to invoke the `circuit2_turtlebot_lidar_qlearn.py` for test, I got the error message below:
```
[2017-10-11 21:10:39,198] Making new env: GazeboCircuit2TurtlebotLidar-v0
Roscore launched!
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
... logging to /home/robot/.ros/log/94144f72-ae85-11e7-99ea-6045cb646ff3/roslaunch-robot-ur-12135.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robot-ur:39569/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [12148]
Gazebo launched!
ROS_MASTER_URI=http://robot-ur:11311/

setting /run_id to 94144f72-ae85-11e7-99ea-6045cb646ff3
process[rosout-1]: started with pid [12171]
started core service [/rosout]
... logging to /home/robot/.ros/log/94144f72-ae85-11e7-99ea-6045cb646ff3/roslaunch-robot-ur-12160.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: turtlebot_gazebo
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share

[rosout-1] killing on exit
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 49, in <module>
    observation = env.reset()
  File "/usr/local/lib/python2.7/dist-packages/gym/core.py", line 104, in reset
    return self._reset()
  File "/home/robot/Documents/workspace/gym-gazebo/gym_gazebo/envs/gazebo_circuit2_turtlebot_lidar.py", line 105, in _reset
    rospy.wait_for_service('/gazebo/reset_simulation')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 159, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown

```
It seems that currently there is no official support for package `turtlebot_gazebo` working under Gazebo8 when using ROS kinetic(The package `ros-kinetic-turtlebot-gazebo` only works for Gazebo7 and it cannot be installed if you've already installed the Gazebo8 followed the guide [here](http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install)).

Does anyone have some idea on how to fix this problem?

vmayoral, 2017-10-11:
Hello @jeasinema, I have an environment with Gazebo 8 installed from the repositories and with `turtlebot_gazebo` compiled from the sources. All working together. Just fetch https://github.com/turtlebot/turtlebot_simulator into your workspace (which contains `turtlebot_gazebo`) and `catkin_make` it.

vmayoral, 2017-10-11:
In other words, even if you can't install it from the repositories you should be able to dump the package into a catkin workspace and compile it by hand.

That'll do. Closing this for now.

closed 2017-10-11 by vmayoral