erlerobot/gym-gazebo #36: CMake error related to

zishanahmed08, 2017-10-11:
It am stuck at the installation at the below step .I get an Cmake error related to ar_track_alvar_msgs package

cd gym_gazebo/envs/installation
bash setup_kinetic.bash		

Below is my log

`CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ar_track_alvar_msgs" with any of the following names:

    ar_track_alvar_msgsConfig.cmake
    ar_track_alvar_msgs-config.cmake

  Add the installation prefix of "ar_track_alvar_msgs" to CMAKE_PREFIX_PATH
  or set "ar_track_alvar_msgs_DIR" to a directory containing one of the above
  files.  If "ar_track_alvar_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  yujin_ocs/yocs_ar_marker_tracking/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/zishanshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/zishanshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
## ROS workspace compiled ##
`

zishanahmed08, 2017-10-11:
Issue resolved with sudo apt-get install ros-kinetic-ar-track-alvar-msgs

closed 2017-10-11 by zishanahmed08

vmayoral, 2017-10-11:
Thanks for reporting it.