erlerobot/gym-gazebo #37: Open Cv error

zishanahmed08, 2017-10-12:
I get the below error when i try a simple turtlebot example after installation.

**ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type**

I tried the 2 solutions mentioned at the below stack overflow link
https://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv

1)Append python path to .bashrc
2)Create symbolic link

Both seems to not work.Also i was confused which path to be used--The dist packages or site packages..So i tried both..But it doesnt work.

Path1- /usr/lib/python3.5/site-packages
Path2- /usr/local/lib/python3.5/dist-packages

OpenCV3 was succesfully installed from sources for Python 3 (git clone https://github.com/Itseez/opencv.git)

vmayoral, 2017-10-12:
@zishanahmed08, that's an identified issue, yes. I also experienced the same when porting `gym-gazebo` to Python 3. Here's what i did:
- remove the malfunctioning `cv2.so` from the path (you can rename it, move it or just remove the corresponding route from `LD_LIBRARY_PATH`
- Compiled cv2.so from sources and installed it (`make install`)

That did it. Note that you need to remove the malfunctioning version of the library from your path.

vmayoral, 2017-10-12:
Closing this for now. Let me know if you experienced further issues.

zishanahmed08, 2017-10-13:
@vmayoral ..Thanks it works.

   
**That did it. Note that you need to remove the malfunctioning version of the library from your path.**
Is there a specific way of doing this or just the renaming on cv2.so at the location /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so ?

zishanahmed08, 2017-12-19:
HI Viktor,
After renaming the malfunctioning cv2.so , i was able to import CV2 on python3 but I get further errors related to cv_bridge on line  107 of gazebo_circuit2c_turtlebot_camera_nn.py i.e **cv_image = CvBridge().imgmsg_to_cv2(image_data, "bgr8")**

The error is as below
**from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost**

I have tried a fresh clean install and documented it here-[Installation_instructions.txt](https://github.com/erlerobot/gym-gazebo/files/1571059/Installation_instructions.txt)
The only change i see is that i pip3 install opencv-python rather than installed from sources.

zishanahmed08, 2017-12-19:
@vmayoral I am unable to re-open the issue.Do i open a new one?

zishanahmed08, 2018-11-21:
I dont remember exactly but its related to the installation procedure because of python2 and python 3. SO i  made a document  of the installation procedure.Just follow it.

https://github.com/erlerobot/gym-gazebo/issues/90

nzlz, 2019-01-15:
Only python2 supported for the Turtlebot Envs for now. Closing.

closed 2019-01-15 by nzlz