erlerobot/gym-gazebo #38: gazebo crashes when running the circuit2_turtlebot_lidar_qlearn example

zishanahmed08, 2017-10-13:
[roslaunch][ERROR] 2017-10-13 13:22:52,241: [gazebo-2] process has died [pid 7674, exit code 255, cmd /home/zishanshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode /home/zishanshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/../assets/worlds/circuit2.world __name:=gazebo __log:=/home/zishanshaikh/.ros/log/d307559c-b008-11e7-a97c-38d547c9c188/gazebo-2.log].

vmayoral, 2017-10-16:
Not enough details provided. @zishanahmed08, **do not write in my personal e-mail and here (twice) if you want to be helped**.

closed 2017-10-16 by vmayoral

zishanahmed08, 2017-10-16:
Hi,
sorry for writing you twice. I can't build the dependencies with `bash setup_kinetic.bash`. I got the following error:

`[ 44%] Building CXX object joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/src/spacenav_node.cpp.o
/home/zishanshaikh/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/joystick_drivers/spacenav_node/src/spacenav_node.cpp:38:19: fatal error: spnav.h: No such file or directory
compilation terminated.
joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/build.make:62: recipe for target 'joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/src/spacenav_node.cpp.o' failed
make[2]: *** [joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/src/spacenav_node.cpp.o] Error 1
CMakeFiles/Makefile2:28612: recipe for target 'joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/all' failed
make[1]: *** [joystick_drivers/spacenav_node/CMakeFiles/spacenav_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j 1" failed
## ROS workspace compiled ##`

I guess this is the issue, why the environment is also not running. Thanks in advance.

vmayoral, 2017-10-16:
Not sure what you're doing and on which system/environment but here's your problem:

> gazebo/gym_gazebo/envs/installation/catkin_ws/src/joystick_drivers/spacenav_node/src/spacenav_node.cpp:38:19: fatal error: spnav.h: No such file or directory

wds2014, 2017-12-17:
try it 
 sudo apt-get install libspnav-dev