erlerobot/gym-gazebo #40: Took over 1000*max_step_size to update a sensor.

AliBaheri, 2017-10-19:
Hi!
When I am trying the scripts for turtlerbot after some episode I am facing with the following message:

`Took over 1000*max_step_size to update a sensor.`

***** Internal Program Error - assertion (diffTime.sec < maxSensorUpdate) failed in void gazebo::sensors::SensorManager::SensorContainer::RunLoop():
/var/lib/jenkins/workspace/gazebo7-debbuilder/build/gazebo-7.3.1/gazebo/sensors/SensorManager.cc(551): Took over 1000*max_step_size to update a sensor.
Aborted
[gazebo-1] process has died [pid 282, exit code 134, cmd /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode /usr/local/gym/gym-gazebo/gym_gazebo/envs/installation/../assets/worlds/circuit2.world __name:=gazebo __log:=/root/.ros/log/1c18d2fa-b4d3-11e7-bb9f-0242ac110002/gazebo-1.log].
log file: /root/.ros/log/1c18d2fa-b4d3-11e7-bb9f-0242ac110002/gazebo-1*.log

vmayoral, 2017-10-19:
@AliBaheri sounds like an issue with your simulation setup/environment. Can you describe in which hardware are you running the simulation? native or through docker?

AliBaheri, 2017-10-19:
Thanks for your prompt response.

Actually I am using docker.
I have followed these steps:

1) `docker pull erlerobotics/gym-gazebo:latest # to fetch the container
# docker build -t gym-gazebo .`

2) `docker run -it erlerobotics/gym-gazebo`

3)`sudo pip install -e [all]`

4) `cd gym_gazebo/envs/installation
bash turtlebot_setup.bash`

5) `d examples/scripts_turtlebot
python circuit2_turtlebot_lidar_qlearn.py`

Is there any missing step?

vmayoral, 2017-10-19:
Looks like you're doing it right. The error that you're experiencing might be related to the use of docker. I'd suggest to try reproducing it natively and report whether you experience the same issues.

closed 2017-10-20 by AliBaheri