erlerobot/gym-gazebo #41: gzclient dose not work during the simulation

AliBaheri, 2017-10-20:
Hi!
I have a question. I am able to run scripts in the example folder, but I am wondering why `gzclient` command dose not work to show how the robot interacts with the world. Thanks for your suggestion.

vmayoral, 2017-10-20:
`gzclient` should show you the robot. Can you dump the output you get when you run any of the examples?

AliBaheri, 2017-10-20:
The command runs without error but no environment is pop out.

vmayoral, 2017-10-20:
Please dump the output of the command line where you launch the script (not the gzclient one).

AliBaheri, 2017-10-20:
when I run the the following command:
`python circuit2_turtlebot_lidar_qlearn.py`

I will get the folloowing outpout:

`/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
[2017-10-20 10:41:16,197] Making new env: GazeboCircuit2TurtlebotLidar-v0
libdc1394 error: Failed to initialize libdc1394
Roscore launched!
... logging to /root/.ros/log/3395e07a-b583-11e7-ad2e-0242ac110007/roslaunch-f62030a9ef2a-192.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
started roslaunch server http://f62030a9ef2a:44095/
ros_comm version 1.11.20

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES

auto-starting new master
process[master]: started with pid [206]
ROS_MASTER_URI=http://f62030a9ef2a:11311/

setting /run_id to 3395e07a-b583-11e7-ad2e-0242ac110007
Gazebo launched!
... logging to /root/.ros/log/3395e07a-b583-11e7-ad2e-0242ac110007/roslaunch-f62030a9ef2a-224.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

process[rosout-1]: started with pid [234]
started core service [/rosout]
started roslaunch server http://f62030a9ef2a:36571/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /usr/local/gym/gy...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[gazebo-1]: started with pid [276]
process[spawn_turtlebot_model-2]: started with pid [277]
process[mobile_base_nodelet_manager-3]: started with pid [286]
[ INFO] [1508496081.584076417]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1508496081.616723429]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
process[cmd_vel_mux-4]: started with pid [291]
process[bumper2pointcloud-5]: started with pid [346]
process[robot_state_publisher-6]: started with pid [397]
[ INFO] [1508496083.037767279, 1.950000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
Unhandled exception in thread started by
sys.excepthook is missing
lost sys.stderr
[spawn_turtlebot_model-2] process has finished cleanly
log file: /root/.ros/log/3395e07a-b583-11e7-ad2e-0242ac110007/spawn_turtlebot_model-2*.log
[ INFO] [1508496083.658357586, 27.490000000]: Physics dynamic reconfigure ready.
[ INFO] [1508496084.069727584, 27.490000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1508496084.102435939, 27.490000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1508496084.178813239, 27.490000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1508496084.179896211, 27.490000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1508496084.198229997, 27.490000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1508496084.243187964, 27.490000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1508496084.244303170, 27.490000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1508496084.252595491, 27.490000000]: Kobuki(ns = //): <tf_prefix> =
[ INFO] [1508496084.253572677, 27.490000000]: Will publish tf. [mobile_base]
[ INFO] [1508496084.277967417, 27.490000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1508496084.291191793, 27.490000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1508496084.346511355, 27.490000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1508496084.383010839, 27.490000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1508496084.420807835, 27.490000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1508496084.429966558, 27.490000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1508496084.447157742, 27.490000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1508496084.460672283, 27.490000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1508496084.460970148, 27.490000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -144     Time: 0:00:09
EP: 2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -42     Time: 0:00:18
EP: 3 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -67     Time: 0:00:27
EP: 4 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.89] - Reward: -69     Time: 0:00:33
EP: 5`

vmayoral, 2017-10-20:
Looks pretty good to me.

`gzclient` should work provided that you're launching it appropriately.

pilifly, 2017-10-20:
Thanks. When/where the command `gzclinet` should run? Should I open a new terminal? 
Pulse, could you please say where the results are save after running the code?

AliBaheri, 2017-10-20:
Thanks. When/where the command gzclinet should run? Should I run it in a new terminal?
Pulse, could you please say where the results are save after running the code?

vmayoral, 2017-10-23:
> Should I run it in a new terminal?

Yes.

> Pulse, could you please say where the results are saved after running the code?

That depends on the script you run and/or the modifications you make to it. Saving the ANNs isn't hard.

closed 2017-11-20 by vmayoral