erlerobot/gym-gazebo #42: Tensorflow conflicts

AliBaheri, 2017-10-20:
Thanks for great works.

I have already locally installed and used Tensorflow and Keras. Also, I am able successfully run the scripts in the examples folder in gym-gazebo via Docker excepts those which has dependensices to Tensorflow. This is really strange, because I have already installed it. I am pretty new to Docker so would you please give some hints in this regard?

`Using TensorFlow backend.
Traceback (most recent call last):
  File "circuit2c_turtlebot_camera_dqn.py", line 16, in <module>
    from keras.models import Sequential, load_model
  File "/usr/local/lib/python2.7/dist-packages/keras/__init__.py", line 2, in <module>
    from . import backend
  File "/usr/local/lib/python2.7/dist-packages/keras/backend/__init__.py", line 64, in <module>
    from .tensorflow_backend import *
  File "/usr/local/lib/python2.7/dist-packages/keras/backend/tensorflow_backend.py", line 1, in <module>
    import tensorflow as tf
ImportError: No module named tensorflow`

AliBaheri, 2017-10-20:
Update: I resolved the issue. But for DQN examples I am facing with the following message:

`roscore cannot run as another roscore/master is already running.
Please kill other roscore/master processes before relaunching.
The ROS_MASTER_URI is http://bccebdd8d756:11311/
The traceback for the exception was written to the log file
... logging to /root/.ros/log/825b759a-b591-11e7-9233-0242ac110002/roslaunch-bccebdd8d756-364.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

sh: 1: killall: not found
sh: 1: killall: not found
root@bccebdd8d756:/usr/local/gym/gym-gazebo/examples/scripts_turtlebot# started roslaunch server http://bccebdd8d756:42137/

SUMMARY`

What are killall? Would you please give some hints?

vmayoral, 2017-10-20:
> Update: I resolved the issue. But for DQN examples I am facing with the following message:

How did you fix it. Can you report it so that other can benefit from your experience?

> What are killall? Would you please give some hints?

That's killing previous instances launched of different processes. You need to do so to relaunch.

> `roscore cannot run as another roscore/master is already running.

This is what happens if you don't.

AliBaheri, 2017-10-20:
I just installed Tensorflow via pip, although I had it already locally, it was strange but worked for me.

Still I do not understand the source of error for DQN. I relaunched, but still DQN not works and SARSA and simple Q learning works well.

AliBaheri, 2017-10-20:
In fact, for several minutes code stucks in the following point:

`Using TensorFlow backend.
[2017-10-20 13:04:18,752] Making new env: GazeboCircuit2TurtlebotLidarNn-v0
libdc1394 error: Failed to initialize libdc1394
Roscore launched!
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
... logging to /root/.ros/log/2f907af8-b597-11e7-919e-0242ac110002/roslaunch-b171ff6f16f3-227.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://b171ff6f16f3:35349/
ros_comm version 1.11.20

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES

auto-starting new master
process[master]: started with pid [241]
ROS_MASTER_URI=http://b171ff6f16f3:11311/

setting /run_id to 2f907af8-b597-11e7-919e-0242ac110002
Gazebo launched!
Traceback (most recent call last):
  File "runtraining_dqn_circuit2_turtlebot_lidar.py", line 275, in <module>
    with open(params_json) as outfile:
IOError: [Errno 2] No such file or directory: '/tmp/turtle_c2_dqn_ep900.json'
... logging to /root/.ros/log/2f907af8-b597-11e7-919e-0242ac110002/roslaunch-b171ff6f16f3-259.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

sh: 1: killall: not found
sh: 1: killall: not found
process[rosout-1]: started with pid [277]
started core service [/rosout]
started roslaunch server http://b171ff6f16f3:34707/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /usr/local/gym/gy...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[gazebo-1]: started with pid [308]
process[spawn_turtlebot_model-2]: started with pid [312]
process[mobile_base_nodelet_manager-3]: started with pid [317]
process[cmd_vel_mux-4]: started with pid [320]
process[bumper2pointcloud-5]: started with pid [321]
[ INFO] [1508504664.460301703]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1508504664.463653854]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
process[robot_state_publisher-6]: started with pid [358]
[ INFO] [1508504664.957947829, 1.390000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1508504665.267378254, 51.620000000]: Physics dynamic reconfigure ready.
[spawn_turtlebot_model-2] process has finished cleanly
log file: /root/.ros/log/2f907af8-b597-11e7-919e-0242ac110002/spawn_turtlebot_model-2*.log
[ INFO] [1508504665.516054391, 51.620000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1508504665.516570370, 51.620000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1508504665.522113464, 51.620000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1508504665.609785904, 51.620000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1508504665.609871274, 51.620000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1508504665.612335963, 51.620000000]: Kobuki(ns = //): <tf_prefix> =
[ INFO] [1508504665.613079272, 51.620000000]: Will publish tf. [mobile_base]
[ INFO] [1508504665.616139692, 51.620000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1508504665.618168932, 51.620000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1508504665.625105866, 51.620000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1508504665.633877693, 51.620000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1508504665.642845675, 51.620000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1508504665.644698284, 51.620000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1508504665.646515662, 51.620000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1508504665.648127431, 51.620000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1508504665.648595744, 51.620000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
'

AliBaheri, 2017-10-20:
Solved.

closed 2017-10-20 by AliBaheri

larenzhang, 2017-11-02:
How did you solve the problem when you stuck in "[ INFO] [1508504665.648595744, 51.620000000]: GazeboRosKobuki plugin ready to go! [mobile_base]"?