erlerobot/gym-gazebo #43: Trouble getting the turtlebot example to run

swordsgnat, 2017-10-20:
I'm attempting to follow along with the Ubuntu 16.04 installation instruction.  I've installed ROS, Gazebo, Python, OpenCV3, all the dependencies, and Sophus; cloned into Gazebo Gym and installed it (is it meant to be cloned into from within the Sophus directory? That seems odd to me, but there's no `cd` line in the installation instructions) ; installed Theano and Keras, and then `CATKIN-IGNORE`d the three directories listed. Then, I tried to run the lines:

```
cd gym_gazebo/envs/installation
bash setup_kinetic.bash	
```
but ran into an error that read:

```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by
"ar_track_alvar_msgs" with any of the following names:

ar_track_alvar_msgsConfig.cmake
ar_track_alvar_msgs-config.cmake
```

which makes it seem like another directory is broken. I `CATKIN_IGNORE`d the `yocs_ar_marker_tracking` and the `yocs_ar_pair_tracking` directories from inside `yujin_ocs`, and then I was able to get a clean and error-less compile from the above lines (well, it still had a `Packages "mav_msgs" not found in the workspace` error but it's always had that and it doesn't seem to be that important). 

Once I got it compiled, I tried to run the first example through:

```
cd gym_gazebo/examples/scripts_turtlebot
python circuit2_turtlebot_lidar_qlearn.py
```

(well, sort of - `examples` is no longer in `gym_gazebo` but is in the outer directory, `gym-gazebo`, but whatever, I just `cd`'d up to it) But when I ran the python line, I got an error that read:

```
Traceback (most recent call last):
File "circuit2_turtlebot_lidar_qlearn.py", line 2, in 
import gym
ImportError: No module named gym
```

which seems kind of fundamental. Forgive me if this is a stupid question (I'm newish to this kind of work) but can anyone give me a tip as to what's going wrong?

vmayoral, 2017-10-23:
> (well, sort of - examples is no longer in gym_gazebo but is in the outer directory, gym-gazebo, but whatever, I just cd'd up to it) But when I ran the python line, I got an error that read:

Thanks for reporting this. Do you want to submit a PR to fix this in the README?

> which seems kind of fundamental. Forgive me if this is a stupid question (I'm newish to this kind of work) but can anyone give me a tip as to what's going wrong?

gym-gazebo is an extension of OpenAI's gym. OpenAI gym needs to be installed. It's a pre-requirement, added it to the README.

swordsgnat, 2017-10-24:
OK, I installed gym, fixed a weird problem I had with my pip, and recompiled everything. Now I have a new error. When I try to run the first example with:

```
cd gym_gazebo/examples/scripts_turtlebot
python circuit2_turtlebot_lidar_qlearn.py
```

I get: 

```
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 3, in <module>
    import gym_gazebo
ImportError: No module named gym_gazebo
```

Looking back through the tutorial, when Gym Gazebo is successfully installed the last message is:

```
  Running setup.py develop for gym-gazebo
Successfully installed gym-gazebo

```

which seems to say that the python module name is "gym-gazebo" not "gym_gazebo" as in the code. But of course, you can't just switch the two in the import statement because python gets angry at you. Am I missing something here, or is this a typo...?

Also, lingering questions from my first post:
1. I had to `CATKIN_IGNORE` `yocs_ar_marker_tracking` and `yocs_ar_pair_tracking` (within `yujiin_ocs`) to get the bash setup file to run. Is that a problem? Are those files necessary? Is there a reason they had a package configuration file error?
2. Should Gym Gazebo be inside the `Sophus` directory, or doesn't it matter?
3. Should examples be inside `gym_gazebo` or is just being in` gym-gazebo` fine? I mean, the whole system should have access to the `gym-gazebo` and `gym` python modules after a proper install, but I have no idea about any other dependencies.

kentsommer, 2017-10-25:
@swordsgnat 

1. You can fix this by editing the ```gazebo.repos``` file which can be found at ```gym_gazebo/envs/installation``` and adding adding the following (in place of the other two alvar related entries)
 ```
  ar_track_alvar:
    type: git
    url: https://github.com/ros-perception/ar_track_alvar.git
    version: kinetic-devel
```

2. While it technically does not matter whether it is inside or not, it should **NOT** be inside the ```Sophus``` folder

3. The examples are in the correct place (the base ```gym-gazebo``` folder)

swordsgnat, 2017-10-25:
Thanks for the answers to the lingering questions, I appreciate it!

Does anyone have any idea about the main question; the `ImportError: No module named gym_gazebo` bug? Like I said above, it seems like the Python module is being installed as `gym-gazebo `instead of `gym_gazebo`? Am I missing something here?

swordsgnat, 2017-10-25:
Update: I've gotten past the gym-gazebo / gym_gazebo error - when I compiled with just `pip` instead of `pip3` (as in `sudo -H  pip install -e .` ) it recognized the module (which seems to imply I'm using Python 2 instead of Python 3, which might have other implications...). Now I have a different error that was raised before in #22, but never answered, I think. This message comes up:

```
/usr/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 164, in make
    return registry.make(id)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 122, in make
    env = spec.make()
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 88, in make
    cls = load(self._entry_point)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 17, in load
    result = entry_point.load(False)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/nate/gym-gazebo/gym_gazebo/envs/turtlebot/__init__.py", line 6, in <module>
    from gym_gazebo.envs.turtlebot.gazebo_circuit2c_turtlebot_camera_nn import GazeboCircuit2cTurtlebotCameraNnEnv
  File "/home/nate/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_circuit2c_turtlebot_camera_nn.py", line 20, in <module>
    import skimage as skimage
ImportError: No module named skimage
```
which seems like some error with a camera data sampler or something? Any ideas?

kentsommer, 2017-10-25:
@swordsgnat 

Just says you are missing scikit-image so install that: 

1. If using pip:
```pip install scikit-image```

2. If using conda:
```conda install scikit-image```

swordsgnat, 2017-10-25:
@kentsommer Thank you!

Ok, I finally got it working. After doing the sudo -H pip install scikit-image thing, I got yet another error, this time the same as the one that got raised in issue #16. The error message looks like:
```
Invalid <arg> tag: environment variable 'GYM_GAZEBO_WORLD_CIRCUIT2' is not set. 

Arg xml is <arg default="$(env GYM_GAZEBO_WORLD_CIRCUIT2)" name="world_file"/>
The traceback for the exception was written to the log file
```

On a hunch, I ran the line `bash turtlebot_setup.bash`  (within gym_gazebo/envs/installation) which isn't mentioned in the installation section (ooh, hey, a new readme!) and then it ran correctly and I got to train the little robot for the first time. Hopefully this is all helpful for anyone else trying to get the example to work. 

Do you guys want me to rewrite the install part so you don't run into the errors I did? (e.g. the ar_track_alvar fix, the scikit-image fix, and this last bash_turtlebot_setup.bash fix, annd clearing up the cds a bit?) I'm willing to do that, I just don't know how universal my errors are / how specific you want the readme to be.

closed 2017-10-25 by swordsgnat

jerpint, 2017-12-28:
@swordsgnat can you please elaborate on your `CATKIN_IGNORE` solution? I am new to ROS and don't really understand where that should go.
I am currently getting the same error you were mentionning when running `bash setup_kinetic.bash` on Ubuntu 16.04. @kentsommer editing `gazebo.repos` did not solve it for me. Here is my error output

```CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "ar_track_alvar_msgs" with any of the following names:

    ar_track_alvar_msgsConfig.cmake
    ar_track_alvar_msgs-config.cmake

  Add the installation prefix of "ar_track_alvar_msgs" to CMAKE_PREFIX_PATH
  or set "ar_track_alvar_msgs_DIR" to a directory containing one of the above
  files.  If "ar_track_alvar_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  yujin_ocs/yocs_ar_marker_tracking/CMakeLists.txt:12 (find_package)

-- Could not find the required component 'ar_track_alvar_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ar_track_alvar_msgs" with any of the following names:

    ar_track_alvar_msgsConfig.cmake
    ar_track_alvar_msgs-config.cmake

  Add the installation prefix of "ar_track_alvar_msgs" to CMAKE_PREFIX_PATH
  or set "ar_track_alvar_msgs_DIR" to a directory containing one of the above
  files.  If "ar_track_alvar_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  yujin_ocs/yocs_ar_marker_tracking/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:290: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
## ROS workspace compiled ##
```

jerpint, 2017-12-28:
EDIT: @kentsommer adding the snippet to `gazebo.repos` yields the following error : 

```CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/_run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/build.make.tmp
CMake Error: : System Error: File name too long
CMake Error: Could not create /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/_run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/cmake_clean.cmake
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/_run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/DependInfo.cmake.tmp
CMake Error: : System Error: File name too long
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/build.make.tmp
CMake Error: : System Error: File name too long
CMake Error: Could not create /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/cmake_clean.cmake
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle.launch.dir/DependInfo.cmake.tmp
CMake Error: : System Error: File name too long
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle_no_kinect.launch.dir/build.make.tmp
CMake Error: : System Error: File name too long
CMake Error: Could not create /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle_no_kinect.launch.dir/cmake_clean.cmake
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_launch_pr2_bundle_no_kinect.launch.dir/DependInfo.cmake.tmp
CMake Error: : System Error: File name too long
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test.dir/build.make.tmp
CMake Error: : System Error: File name too long
CMake Error: Could not create /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test.dir/cmake_clean.cmake
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test.dir/DependInfo.cmake.tmp
CMake Error: : System Error: File name too long
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_param_config-basic.test.dir/build.make.tmp
CMake Error: : System Error: File name too long
CMake Error: Could not create /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_param_config-basic.test.dir/cmake_clean.cmake
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws [... cut ...]
CMake Error: Cannot open file for write: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build/ar_track_alvar/ar_track_alvar/CMakeFiles/run_tests_ar_track_alvar_roslaunch-check__home_jeremy_gym-gazebo_gym_gazebo_envs_installation_catkin_ws_src_ar_track_alvar_ar_track_alvar_test_marker_param_config-full.test.dir/progress.make.tmp
CMake Error: : System Error: File name too long
-- Build files have been written to: /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/build
Invoking "cmake" failed
```

zishanahmed08, 2017-12-29:
just install it with the belwo command
 sudo apt-get install ros-kinetic-ar-track-alvar-msgs

jerpint, 2017-12-29:
@zishanahmed08 thanks, I was missing a lot of other dependencies as well. I've sucesfully been able to compile the ros workspace. However, I currently run in to this error which I haven't been able to get around after running `python3 circuit2_turtlebot_lidar_qlearn.py` :

```while processing /home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/jeremy/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot/turtlebot_description/robots/kobuki_hexagons_asus_xtion_pro.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(arg urdf_file)" name="robot_description"/>
The traceback for the exception was written to the log file```

any ideas as to what might be causing this/ how to fix it?

zishanahmed08, 2017-12-29:
Refer the issue at below link of closed issue 58
https://github.com/erlerobot/gym-gazebo/issues/58