erlerobot/gym-gazebo #45: Stuck in yujin_ocs part when "bash setup_kinetic.bash" in Ubuntu16.04

NanxunDuo, 2017-10-24:
I am successfully go through all the steps before "bash setup_kinetic.bash", and I have used _**sudo apt-get install ros-kinetic-yujin-ocs**_ to install yujin-ocs and it still went wrong like this
 
[image]

Anyone have ideas about solving this?
Many thnks!!

swordsgnat, 2017-10-24:
I had a problem with the same package - it seems like something's just not right with it. That said, it also seems pretty teriatry. I `CATKIN_IGNORE`d `yocs_ar_marker_tracking` and `yocs_ar_pair_tracking` within `yujin_ocs` the same way you have to for the other three packages in the tutorial, and then the bash ran properly. 

Grain of salt, because I still haven't gotten the tutorial to work (currently stuck on a "no module named gym_gazebo" bug) but that might help you past that part.

kentsommer, 2017-10-25:
@NanxunDuo 

Your actual issue is higher up in the build. Please scroll up and look for the first error.

NanxunDuo, 2017-11-01:
@kentsommer 
[image]

This is my very first error running bash setup_kinetic.bash.
And the last error has changed to lack of packages tf2_sensor_msgs
[image]
So could you please help me to tell what is the exact error?

kentsommer, 2017-11-01:
@NanxunDuo 

1. You can fix these errors by editing the gazebo.repos file which can be found at gym_gazebo/envs/installation and adding adding the following (in place of the other two alvar related entries)
```
ar_track_alvar:
   type: git
   url: https://github.com/ros-perception/ar_track_alvar.git
   version: kinetic-devel
```

2. Also add this to the above file in place of the other mav_comm entry
```
mav_comm:
   type: git
   url: https://github.com/ethz-asl/mav_comm.git
   version: master
```

closed 2017-11-20 by vmayoral