erlerobot/gym-gazebo #46: about "tf2_sensor_msgs" when run bash setup_kinetic.bash

larenzhang, 2017-10-27:
My development envirament is ubuntu 16.04.
After install ROS and gym-gazebo,I try to install turtlebot.But when I run bash setup_kinetic.bash ,there is a problem as follows:
-- Could not find the required component 'tf2_sensor_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf2_sensor_msgs"
  with any of the following names:

    tf2_sensor_msgsConfig.cmake
    tf2_sensor_msgs-config.cmake

  Add the installation prefix of "tf2_sensor_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_sensor_msgs_DIR" to a directory containing one of the above files.
  If "tf2_sensor_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  pcl_ros/pointcloud_to_laserscan/CMakeLists.txt:4 (find_package)

And I try to solve this problem according to https://answers.ros.org/question/202078/unable-to-install-pointcloud_to_laserscan-ros-indigo/.I try to install pointcloud_to_laserscan-ros-indigo by running sudo apt-get install ros-indigo-pointcloud-to-laserscan,but the command line returns "E: Unable to locate package ros-indigo-pointcloud-to-laserscan".

Do you have the same experience?Please give me some suggestions.Thank you!

larenzhang, 2017-10-28:
I solved the problem by installing ros-kinetic-tf2-sensor-msgs .

closed 2017-10-28 by larenzhang

NanxunDuo, 2017-12-01:
@larenzhang Hi Laren! 
 I also meet the same problem with 

> CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf2_sensor_msgs"
  with any of the following names:

    tf2_sensor_msgsConfig.cmake
    tf2_sensor_msgs-config.cmake

  Add the installation prefix of "tf2_sensor_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_sensor_msgs_DIR" to a directory containing one of the above files.
And when I try 

> sudo apt-get install ros-kinetic-tf2-sensor-msgs

I failed with
[image]

Have you ever meet this before? I try to find libicu55.1-7ubuntu0.2  but all I found is libicu55.1-7 and libicu55.1-7ubuntu0.3.
I installed libicu55.1-7ubuntu0.3 and tried to install tf2-sensor-msgs again and the error remain the same.
Do you have any idea on this?Thanks a lot!
Regards,
Elizabeth