erlerobot/gym-gazebo #47: Sarsa and DQN Example Issues - Slow Turning

swordsgnat, 2017-10-27:
I've gotten the Turtlebot Q learning example to work, but when I try to run the Sarsa or DQN examples the Turtlebot stays on the spawn point without advancing and just turns excruciatingly slowly. Does anyone know why this is happening? It only turns one direction (right) and at what seems like a constant rate.

kentsommer, 2017-11-01:
@swordsgnat 

This implies that the "training" script crashed. Scroll near the top after starting training and look carefully for an error. You will find one :)

swordsgnat, 2017-11-10:
Ok, I went and fixed the erros arising from Python3/Python2 discrepencies, and now DQN and Sarsa work, but I still get this spinning error. Every couple times I run the training process, after a few hundred epochs, the turtlebot will stop moving wherever it is and just slowly spin. I agree that this must mean the training script has failed (the automatic timeout I put in doesn't trigger, for one) but there's no printed errors whatsoever. 

Does anyone have any idea what might be causing this and how to fix it?

swordsgnat, 2017-11-15:
@kentsommer 

I've been continuing to work with the program, and these errors keep regularly cropping up. There seem to be two kinds - in one, the robot gets "stuck" and stops moving forward, but the simulation doesn't reset and the rewards keep compounding. For that one, if I manually move the robot in the Gazebo window or reset the simulation, the training will continue as intended. That error normally happens in a corner. In the other kind, the robot will stop moving wherever it is (I can't seem to find a location-dependence) and just spin very _very_ slowly. Moving the robot manually or resetting the scenario doesn't help - the robot will be moved / respawn to the start, but it won't move forward again and the training will not continue. In either kind, there are **no error messages whatsoever** in the entire readout. I agree that it looks like the training script is failing, but I have no idea where, how, or why. 

I can work around the first kind programmatically, but the second kind is insidious - it prevents me from training more than about 1000 iterations. That prevents me from seeing real convergence, which is the crucial factor I'm looking for. 

Does anyone have any ideas what this could be or how to fix it?

kentsommer, 2017-11-15:
@swordsgnat 

1. I'm not sure what you mean by "stuck", the simulation should reset if the laser readings are below a threshold. So the only reason it wouldn't reset after a loop is because that condition is never being met. 

2. The second one is caused by a bug that is STILL present in the gazebo 7 deb install :-1: What happens is that there is an assert that is hard coded to check that the sensor update took less than 1 second. But when gazebo is running at > real-time it triggers eventually and crashes the simulation. 

Luckily... you can install gazebo from source instead and patch it yourself!

Source install instructions: http://gazebosim.org/tutorials?cat=install&tut=install_from_source&ver=7.0

Patch details (simply edit the source files to reflect these changes before building gazebo and installing): https://bitbucket.org/osrf/gazebo/pull-requests/2115/update-sensor-assert/diff

closed 2017-11-20 by vmayoral

chapulina, 2017-12-08:
Note that the pull request mentioned above has been released on Gazebo 7.1.0 on April 2016, just setup your computer to accept deb packages from OSRF as described on [this](http://gazebosim.org/tutorials?cat=install&tut=install_ubuntu&ver=7.0#Alternativeinstallation:step-by-step) tutorial and you should get the latest Gazebo 7 version (currently 7.9). No source install needed :wink:

swordsgnat, 2017-12-08:
Thing is, that _is_ how I installed Gazebo originally - I've only been using it since September this year, version 7.8. When I tried the source install, the lines were already fixed. I still get the stop-and-slow-turning error cropping up and messing up my training. Could there be something else causing the problem, since this fix appears to not solve it?

pulver22, 2018-11-06:
Hi @swordsgnat, were you able to solve this problem? I get the same, actually from the first episode while running `python3 circuit_turtlebot_lidar_qlearn.py`.

swordsgnat, 2018-11-06:
No, I never did - I just attempted enough training that eventually one set went for a few thousand episodes, and that was good enough for what I was working on. I haven't touched the project in months, now.

pulver22, 2018-11-07:
Mmm ok, unfortunately I have the weird behaviour just as soon as I spawn the robot in Gazebo. So there is no chance to get a small learning at all.