erlerobot/gym-gazebo #48: AbAout "ImportError: No module named PyQt4.uic.pyuic" when run "bash setup_kinetic.bash"

larenzhang, 2017-10-28:
On ubuntu 16.04,when I run bash setup_kinetic.bash and it cmake about 98%,the program stuck into error as follows:
Traceback (most recent call last):
  File "/usr/bin/pyuic4", line 3, in <module>
    import PyQt4.uic.pyuic
ImportError: No module named PyQt4.uic.pyuic
kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite.dir/build.make:57: recipe for target 'kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite' failed
make[2]: *** [kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite] Error 1
CMakeFiles/Makefile2:54244: recipe for target 'kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite.dir/all' failed
make[1]: *** [kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j 1" failed
## ROS workspace compiled ##

Maybe it's about python version.I'm not sure.Does anybody have ideas?

swordsgnat, 2017-10-30:
That looks like an error from "kobuki_qtestsuite" which is one of the packages they list in the installation guide as broken for some reason. If you make a file just titled "CATKIN_IGNORE" inside that folder (which you can find inside the catkin_ws/src/kobuki_desktop path inside gazebo-gym iirc) then catkin will skip over that and move on, which is fine because it's tertiary. 

The link here provides more info on `CATKIN_IGNORE`ing stuff. You're probably gonna have to do this to the `wiimote`, `yocs_ar_pair_tracking`, `yocs_ar_marker_tracking`, and `spacenav_node` folders too. Let me know if you need more help with this part!
https://answers.ros.org/question/208180/how-to-use-catkin_ignore-file-correctly/

kentsommer, 2017-11-01:
@swordsgnat @larenzhang 

This does not require ignoring the package. Simply install the missing component:

`sudo apt-get install python-qt4`

larenzhang, 2017-11-01:
I have installed python3-qt4 before run the command "bash setup_kinetic.bash".And then I have installed python-qt4 according to @kentsommer .
However,there is still a error.It seems that the error is "ImportError: No module named PyQt4.uic.pyuic" when I import PyQt4.uic.pyuic in the environment of python2.7 and the error is "Error: one input ui-file must be specified" in the environment of python3.5.I find that the compile environment of  catkin_ws is python2.7.
Finally,I solved it by modifying the code "import PyQt4.uic.pyuic" as "#import PyQt4.uic.pyuic" in the file /usr/bin/pyuic4 and the program compiled successfully.I think the file /usr/bin/pyuic4 just has one sentence and it won't matter other codes.
My solution is not elegant.I don't now whether there are other influences.Maybe someone can come up with better solutions.
@swordsgnat @kentsommer Thanks for your kind-hearted.

kentsommer, 2017-11-01:
@larenzhang 

Can you open a new python shell and try `import PyQt4` and post the output if there are any errors?

larenzhang, 2017-11-01:
I import PyQt4 in both python2.7 and python3.5 shell.It is ok in python3.5.There is a error in python2.7 and the error is "ImportError: No module named PyQt4".

kentsommer, 2017-11-01:
@larenzhang 

Well if its not loading in 2.7 that is why you can't build the `kobuki_qtestsuite` package.

larenzhang, 2017-11-01:
Do you know how to deal with this problem?I have installed python-qt4.

kentsommer, 2017-11-01:
@larenzhang 

Try uninstalling `python3-qt4`

vmayoral, 2017-11-20:
@larenzhang i'd suggest focusing only in Python 3.

Closing (thanks @kentsommer for the support). Feel free to re-open it if necessary.

closed 2017-11-20 by vmayoral