erlerobot/gym-gazebo #49: The example agent pause running stangely!

larenzhang, 2017-11-02:
There are main two questions when I run the example code.
1.It occured that the program paused strangely almost everytime I run circuit2_turtlebot_lidar_qlearn.py.
And there is no error output.The occasion is as follows:
[image]
2.It seems everything ok and there is no error output when I run circuit2_turtlebot_lidar_dqn.py,but the robot don't move and the program seems to wait something,the output as follows:
[image]

Have you ever encountered these problems?

larenzhang, 2017-11-03:
I have solved the second question.The reason causing the error is that there are some python2 grammer in circuit2_turtlebot_lidar_dqn.py.I reedit the file to python3 and now it works well.
However,it's hard to converge when I train DQN with just lidar data as inputs.Maybe the hyper-parameters are not quite correct.
Have you train DQN well to solve the enviroment?And could you please offer the worked hyper-parameters?

swordsgnat, 2017-11-13:
I've been using DQN training, and I get kind of a sine wave of results. The 'bot will get really good after iteration 100 or so, stay good for like 50 iterations, and then tank for the next hundred or two. Then it'll get good again, etc. I've been playing with the epsilon exploration rate decay to see what that did - cutting the exploration rate to zero after the first iteration scoring over, say, 2000, has had a good effect on the quality of the runs afterwards, but the 'bot still fails early about every four iterations or so. I'm still testing / playing with it (and I haven't even started playing with the rewards yet) but I'll let you know if I find anything.

vmayoral, 2017-11-20:
> @rkojcev @vmayoral @ahcorde @imuguruza @jlamperez
Please do not cite us all. It's a really, really bad practice if you intend to get support from us.

@larenzhang, can you describe your first question in more detail? Can you describe in which system you're currently running the gym? have you tried debugging the environment you're running?

Maclapa, 2017-12-05:
Hi, I have the same problem as you described. In my case the program pause when it runs for higher number of episodes. The robot in simulation is constantly very slowly moving but I think this is the case of friction of environment.
[image]
[image]

waply, 2018-01-19:
@swordsgnat  
Hi, I have the same problem as you described。 the robot gets "stuck" and stops moving forward。
Does anyone have any ideas what this could be or how to fix it?

zengjie617789, 2018-05-08:
@swordsgnat @larenzhang 
 hi! i have been using the DQN too. but as the same problem "The example agent pause running strangely! "  is there anyone can help to solve these problem?
[image]

pulver22, 2018-11-06:
Hi @Maclapa did you understand why the gazebo continuously stopped and the robot moves really slow? I have the same exact behaviour.

nzlz, 2019-01-15:
This issue is a mix of different issues, @pulver22  please open a new **Bug report** to facilitate issue-tracking. We will be happy to help you.

closed 2019-01-15 by nzlz