erlerobot/gym-gazebo #53: The need to run one simulation step for each step

Superchicken1, 2017-11-16:
Hi, i have a more theoretical question.
I am looking at the turtle bot example and i also want to do Reinforcement Learning using Gazebo and Ros. As far i understood it does unpause the simulation, publishes a command, waits for its observation and pause again.
In the gym environments of OpenAi they are doing one frame step for each step. So they are performing one action, do one simulation step and get their observation.

How can we be certain to have the right timing for our observation? Isn't it possible that the time between unpause and pause varys for each iteration?

greetings and thx for the awesome work

vmayoral, 2017-11-20:
Hello @Superchicken1, 

> In the gym environments of OpenAi they are doing one frame step for each step. So they are performing one action, do one simulation step and get their observation.

This is not necessarily true. To our understanding, OpenAI's gym does operate that way (step through action-step-observations) in all environments. Particularly, it operates differently working with robot-related environments (using Mujoco). To my understanding, they run the simulation continuously taking actions, observations and samples on the way. No "one frame step for each step".

> How can we be certain to have the right timing for our observation? Isn't it possible that the time between unpause and pause varys for each iteration?

That's a great question. Newer environments just [sleep](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/modular_scara/gazebo_modular_scara_3dof_v2.py#L543) until the action has been performed at the `slowness` rate. We've experimentally tested this and it works pretty decently for reasonable `slowness` rates (1-10 Hz) however it gets messy when going to much higher rates. We are also considering using `moveit` and ROS services to get certainty about actions being completed before continuing with the simulation but that'll happen (if it does) in the future.

This is currently our best answer to your "one frame step for each step". Another approach (which we internally call "continuous running") for the same environment is implemented [here](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/modular_scara/gazebo_modular_scara_3dof_v3.py). This resembles much more to what we understand, OpenAI is proposing for their robot envs. 

@rkojcev, would you like to add something?

rkojcev, 2017-11-20:
@vmayoral I agree with your answers and it is a very good description of our findings regarding the behavior of the continuous action space algorithms. 

In our experiments, the continuous action space algorithms perform as Victor described (continuous sampling during execution of an action).

Superchicken1, 2017-11-21:
@vmayoral and @rkojcev 
Thank you for your Answers.

> This is not necessarily true. To our understanding, OpenAI's gym does operate that way (step through action-step-observations) in all environments. Particularly, it operates differently working with [...]

I am currently trying out mujoco and I think I found something.

You have to pass a frame_skip parameter on the MujocoEnv creation (see [mujoco_env.py](https://github.com/openai/gym/blob/master/gym/envs/mujoco/mujoco_env.py)).
There is also a important "timestep"-Varible in the option tag of the mujoco MJCF-xml files (like [here](https://github.com/openai/gym/blob/master/gym/envs/mujoco/assets/swimmer.xml)). This defines the simulation time step in seconds. 
With both variables are also used to calculate the time delta per step:`def dt(self): return self.model.opt.timestep * self.frame_skip` so i think it is pretty accurate.

So as far i understand it is not doing actions, observations and samples on the way. It is pausing after each step.

> That's a great question. Newer environments just sleep until the action has been performed at the slowness rate. We've experimentally tested this and it works pretty decently for reasonable slowness [...]

I also tried out 'rospy.sleep(timeforaction)' in my ros-gazebo-env step function. Note that rospy.sleep(x) sleeps for x seconds in simulation time.
Looks like this:
```
_step(self, action):
   unpause()
   publish(action)
   rospy.sleep(1.0)
   getObservation()
   pause()
   ...
```
But if I think about it that doesn't solve the problem. There is still time variance before and after the sleep. It just reduces the amount of steps till the action is performed. Also think that this prevents small and short interactions with the environment.
I still think this is problematic if agent doesn't get the right observations on his actions because of the time differences. Maybe your experiments ran on real time so that the "ROS-delay" was not that significant?

Sooo...
I think the only solution is a do-x-simulation-steps or run-simulation-for-x-secounds Ros-service. Is this what you mean with `moveit`? I read that such a command is possible in Gazebo and could be possible with Ros with plugins.
I know this all sounds a bit petty, but I had problems with Ros-Gazebo and I think all this might be the reason. Could it be true that preciser observations lead to a higher sample efficiency?

Greetings and thx again for the awesome project and your time reading this post :-)

vmayoral, 2017-12-02:
Hello @Superchicken1 
> I think the only solution is a do-x-simulation-steps or run-simulation-for-x-secounds Ros-service. Is this what you mean with moveit? I read that such a command is possible in Gazebo and could be possible with Ros with plugins.

That's exactly right. Using `moveit` would be a way to improve the existing environment.

Right now, we don't have a better answer for you. We're currently using this setup and it's working pretty fine to do simulation-to-real-robot training but certainly, there are quite a few aspects that could be researched in here. It'll be a nice contribution. We have some available funding to look into problems like this. Would you be interested in exploring this yourself?

You can write me at victor at erlerobotics.com if that's the case.

Superchicken1, 2017-12-28:
Hi 
Thank you for the offer. But I already moved on to mujoco. 

Greetings and thx for the awesome project.

closed 2017-12-28 by Superchicken1