erlerobot/gym-gazebo #54: How to reset the initial pose of turtlebot at model spawn after every episode/epoch

zishanahmed08, 2017-11-18:
I would like to set the yaw of the turtlebot spawned after every episode/epoch as this would lead to better learning since the turtlebot explores in 360 degrees instead of exploring just ahead.
Environment used -GazeboCircuit2cTurtlebotCameraNnEnv-v0

To do this i tried to edit the **spawn_model** script located at below location
../gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts

I edited the yaw value as below, where the 3rd element of self.initial_rpy  is y(yaw)

> self.initial_rpy               = [0,0,random.uniform(0, 2*math.pi)]# randomly set yaw from 0 to 360 degrees
> earlier it was 
> self.initial_rpy               = [0,0,0]

So at runtime the env calls the below mentioned kobuki.launch.xml and it spawns the turtlebot_model with random yaw.
../gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml

But the problem is this happens only once at the creation of the launch
spawn_turtlebot_model(gazebo_ros/spawn_model)
And i need the yaw to be set randomly at every iteration after the turtlebot  hits the wall and resets. 
I reset the simulation at every iteration with  **rospy.ServiceProxy('/gazebo/reset_simulation', Empty)**
I additionally  reset the world  at every iteration with **rospy.ServiceProxy('/gazebo/reset_world', Empty)**
This did not help

vmayoral, 2017-11-20:
Check [this environment](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/modular_scara/gazebo_modular_scara_3dof_v2.py). Implemented for another robot however it should give you an intuition. You should be able to see there how the initial pose of the SCARA arm is set every time `reset()` gets called.

closed 2017-11-20 by vmayoral

zishanahmed08, 2017-11-23:
@vmayoral Thanks a lot! II was able to do it with topic /gazebo/set_model_state.

NanxunDuo, 2017-12-01:
@zishanahmed08 Hi Zishan! I also try to set initial pose of turtlebot after every episode. Can I ask how to do it through topic /gazebo/set_model_state specifically? By calling service  /gazebo/set_model_state? And which file do you edit? Is it **gazebo_circuit.py** under **~/gym_gazebo/envs/turtlebot** or **circuit_turtlebot.py** under **~/example/turtlebot?** 
Many thanks!

zishanahmed08, 2017-12-01:
Hi,
**Can I ask how to do it through topic /gazebo/set_model_state specifically? By calling service /gazebo/set_model_state?** 
Yes
**And which file do you edit? Is it gazebo_circuit.py under ~/gym_gazebo/envs/turtlebot or circuit_turtlebot.py under ~/example/turtlebot?**
gazebo_circuit.py under ~/gym_gazebo/envs/turtlebot

NanxunDuo, 2017-12-11:
@zishanahmed08 Thank you for your answer! It really helped!