erlerobot/gym-gazebo #55: Install process for 16.04

StevieZissou, 2017-11-18:
Hi sorry if obvious question but for the Ubuntu 16.04 install routine , should I install basic requirements below manually and then proceed to the dependency install section ? It looked like on the 14.04 install there was some special step to remove the ROS Gazebo package and install later version.
Many thanks in advance
---
Basic requirements:

    ROS Kinetic (/rosversion: 1.12.7)
    Gazebo 8.1.1
    Python 3.5.2
    OpenCV3, installed from sources for Python 3 (git clone https://github.com/Itseez/opencv.git)
    OpenAI gym

vmayoral, 2017-11-20:
@StevieZissou, yes, install them manually.

closed 2017-11-20 by vmayoral

jacknlliu, 2017-12-07:
@vmayoral  Why not gazebo7 which shipped with ROS kinetic?