erlerobot/gym-gazebo #57: Simulation stops because of that it's unable to communicate with ROS Master

Aaron9477, 2017-11-21:
Hi, I have a problem with simulation.
My environment is that ubuntu16.04 and gazebo8. When I run the example, after hundreds of times simulation, it will have a warning about 'unable to communicate with ROS Master, registrations are now out of sync', which I can't find out the reason. With ros, I worked well with other package. Is that I run simulation for too long, to cause this problem?
Waiting for your reply! Thank you!
[image]

vmayoral, 2017-11-21:
Thanks for reporting @Aaron9477. Can you provide more details about your system? Specs?

Have you checked if the roscore process remains active or if it gets killed? Is this behavior deterministic? What's your memory state? Have you tried reproducing this behavior in another machine?

Aaron9477, 2017-11-22:
Hi @vmayoral !
Thanks for your reply!
I use Ubuntu16.04, and my computer system has 16G memory and NVIDIA graphics 1080.
Now we suspect that maybe because of connecting with wifi, and the IP of my computer sometimes changes. Then, the roscore no responses. We will have an another test this day with no connection with Internet. I will send my result when we get it.

vmayoral, 2017-11-22:
> Now we suspect that maybe because of connecting with wifi, and the IP of my computer sometimes changes. Then, the roscore no responses. We will have an another test this day with no connection with Internet. I will send my result when we get it.

That sounds about right. I'll keep the issue open for a while.

Aaron9477, 2017-11-23:
Hi, @vmayoral 
After our test, this issue is absolutely caused by the change of IP. After fixing the IP, the simulation works well. I am sorry to bother you. Thanks for your time!

closed 2017-11-23 by Aaron9477

vmayoral, 2017-11-23:
Glad it got solved. Cheers.