erlerobot/gym-gazebo #58: Running an example turtlebot script returns invalid <param>  tag error

harish1696, 2017-11-25:
When I run 'python circuit2_turtlebot_lidar_qlearn.py ' it returns 

while processing /home/harish/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/harish/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot/turtlebot_description/robots/kobuki_hexagons_asus_xtion_pro.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(arg urdf_file)" name="robot_description"/>
The traceback for the exception was written to the log file

harish1696, 2017-11-26:
The issue got solved by replacing '
$(find xacro)/xacro --inorder  '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'
with
$(find xacro)/xacro.py  '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'

in kobuki.launch.xml file in /gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes directory.

harish1696, 2017-11-26:
But I'm not able to see the turtlebot in the simulation?

carolc0708, 2017-11-26:
`vim ~/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/turtlebot_simulator/turtlebot_gazebo/launch/includes/kobuki.launch.xml`
 replace line6 to 
 `<arg name="urdf_file" default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'"/>`
then the problem might be solved

closed 2017-11-28 by harish1696

HareshMiriyala, 2019-03-01:
@carolc0708  @harish1696  Thanks guys !