erlerobot/gym-gazebo #59: KeyError: 'ROS_PORT_SIM'

borongyuan, 2017-11-27:
When I tried to run the example, I got the following error. This appears on both 14.04 and 16.04.

Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 164, in make
    return registry.make(id)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 122, in make
    env = spec.make()
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 89, in make
    env = cls(**self._kwargs)
  File "/home/borong/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_circuit2_turtlebot_lidar.py", line 20, in __init__
    gazebo_env.GazeboEnv.__init__(self, "GazeboCircuit2TurtlebotLidar_v0.launch")
  File "/home/borong/gym-gazebo/gym_gazebo/envs/gazebo_env.py", line 20, in __init__
    port = os.environ["ROS_PORT_SIM"]
  File "/usr/lib/python2.7/UserDict.py", line 23, in __getitem__
    raise KeyError(key)
KeyError: 'ROS_PORT_SIM'

pmwenzel, 2017-11-28:
The function gazebo_env.py changed. You have to the the env vaiable ROS_PORT_SIM in your cmd line before executing the script. 
`export ROS_PORT_SIM=11311` (default port)
Hope that helps.

borongyuan, 2017-11-28:
Thanks @pmwenzel

closed 2017-11-28 by borongyuan