erlerobot/gym-gazebo #60: [gazebo-1] process has died after executing turtlebot example.

Maclapa, 2017-11-30:
Hi, I get an error message after execution of circuit2_turtlebot_lidar_qlearn.py. There is as well no output file in outdir.
[image]
The process of learning runs normally and there is no problem with running gzclient. I do not know what is the reason of error.

aalexsmithh, 2017-12-01:
I'm having a similar issue. 

I don't know if I'd call it a problem though. Gazebo crashes when your python script ends. I'm not sure if this is intentional in the `GazeboEnv.close()` method, or if this is a bug. 

At least in my implementation, all the learning actually gets done before Gazebo crashes at the end of the script. What's probably happening in that script is that the logs are not written before `env.close()` is called, so they just never get written, although that's just my suspicion. Best of luck debugging!

aalexsmithh, 2017-12-01:
Just took a look.

As of commit [`a29cac4`](https://github.com/erlerobot/gym-gazebo/commit/a29cac4995025abf2917c32834bdfb4680b94676), [line 28](https://github.com/erlerobot/gym-gazebo/blob/2dc5eb35d77a48043232a1768e9de6919eecee23/examples/turtlebot/circuit2_turtlebot_lidar_qlearn.py#L28) of that file is commented out, so nothing will be written to `outdir`.

Try uncommenting line 28 and 29. Otherwise the crashing of gazebo seems to be normal behaviour.

Maclapa, 2017-12-04:
Hi,
Thank you for your reply. Unfortunately when the lines are uncommented the robot does not start learning process. It is slowly moving but I think it is caused by the small friction of objects. I think this may be problem is similar to this issue #49 but the pause is on the first step.
[image]

Brudasoe, 2017-12-04:
The main culprit seems to be [line 28](https://github.com/erlerobot/gym-gazebo/blob/2dc5eb35d77a48043232a1768e9de6919eecee23/examples/turtlebot/circuit2_turtlebot_lidar_qlearn.py#L28)
This syntax is deprecated (via [official gym documentation](https://github.com/openai/gym#what-s-new))
Correct version should be: ` env = gym.wrappers.Monitor(env, outdir, force=True)`
Now the simulation should start normally.

Maclapa, 2017-12-04:
Thank you Brudasoe, you are right! 
Now it works perfectly!

closed 2017-12-04 by Maclapa