erlerobot/gym-gazebo #61: Process has died

redman0226, 2017-12-03:
<pre>
[gazebo-2] process has died [pid 14768, exit code -9, cmd /home/redman/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode /home/redman/gym-gazebo/gym_gazebo/envs/installation/../assets/worlds/circuit2c.world __name:=gazebo __log:=/home/redman/.ros/log/81d099b8-d830-11e7-9a76-6045cba8caed/gazebo-2.log].
log file: /home/redman/.ros/log/81d099b8-d830-11e7-9a76-6045cba8caed/gazebo-2*.log
[master] process has died [pid 15286, exit code -9, cmd rosmaster --core -p 11311 -w 3 __log:=/home/redman/.ros/log/173ad282-d833-11e7-9a76-6045cba8caed/master.log].
log file: /home/redman/.ros/log/173ad282-d833-11e7-9a76-6045cba8caed/master*.log
</pre>

Process has died
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

After we run this instruction "python circuit2c_turtlebot_camera_dqn.py ", we have the strange problem,
and have no idea for a long time . 

1.We can't find "gazebo-2.log" .
2.How to fix "the process has died".
Thanks.

vmayoral, 2017-12-03:
Hello @redman0226, if I'm not mistaken, that terminal dump happens when the simulation has finished and gazebo is closed forcefully.

If you'd prefer a "cleaner finishing" I'd suggest looking at the corresponding functions (I'd start looking [here](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_env.py)). A pull request will be more than welcome btw!

closed 2017-12-03 by vmayoral

redman0226, 2017-12-07:
Thanks. It can works.
Now, I can run circuit2_turtlebot_lidar_qlearn.py.
However, I want to run circuit2c_turtlebot_camera_dqn.py, but I open the gazebo and the turtlebot is still at the original position, it didn't start to drive.

The following steps are:
1. bash turtlebot_setup_cnn.bash
2. python circuit2c_turtlebot_camera_dqn.py

Am I missing any steps to run?
Thank you.