erlerobot/gym-gazebo #65: KeyError: "ROS_PORT_SIM"

jamalavedra, 2017-12-10:
hi,

I'm trying to compile the Demo of the 'circuit2_turtlebot_lidar_qlearn.py' but I'm getting a Key Error.
This error is in the code file "gazebo_env.py" and occurs in the line <port=os.environ["ROS_PORT_SIM"]>.
As this file is present in all the presented examples I can't run any of them. 
Any idea on how to fix this?
thanks
Jaume

zishanahmed08, 2017-12-10:
Refer the solution of issue 59 at below link

https://github.com/erlerobot/gym-gazebo/issues/59

jamalavedra, 2017-12-10:
thank you, that worked.

Any idea on this error that I get now?
Invalid <arg> tag: environment variable 'GYM_GAZEBO_WORLD_CIRCUIT2' is not set. 

Arg xml is <arg default="$(env GYM_GAZEBO_WORLD_CIRCUIT2)" name="world_file"/>
The traceback for the exception was written to the log file

jamalavedra, 2017-12-11:
fixed that by using <export GYM_GAZEBO_WORLD_CIRCUIT2=home/user/gym-gazebo/gym_gazebo/envs/assets/worlds/circuit2.world>

closed 2017-12-30 by vmayoral