erlerobot/gym-gazebo #66: [ERROR] [1512985229.869755232]

jamalavedra, 2017-12-11:
Skipping XML Document "/opt/ros/kinetic/share/gmapping/nodelet_plugins.xml" which had no Root Element.  This likely means the XML is malformed or missing.

This is the message what I get when executing the 'circuit2_turtlebot_lidar_qlearn.py'. 

Any idea on how to fix this error?
thanks
jaume

vmayoral, 2017-12-30:
@jamalavedra did you overcome this problem?

Brudasoe, 2018-01-16:
This is problem associated with ROS gmapping package, as mentioned in [this issue](https://github.com/ros-perception/slam_gmapping/pull/56) if the problem persist, I think updating packages or manually copying the [nodelet plugins file](https://github.com/ros-perception/slam_gmapping/blob/hydro-devel/gmapping/nodelet_plugins.xml) will solve it.

nzlz, 2019-01-15:
Seems like the problem was solved thanks to @Brudasoe , if not, please feel free to open a new **Bug report** issue to facilitate issue-tracking.

closed 2019-01-15 by nzlz