erlerobot/gym-gazebo #67: How to add extra model in gym-gazebo environment?

NanxunDuo, 2017-12-12:
I am trying to add some cubes as obstacles when turtlebot moving in environment GazeboCircuit2TurtlebotLidar-v0. And in every episode cubes randomly change their position.
I try to add one cube.
My plan is to edit _gazebo_circuit2_turtlebot_lidar.py_, and edit **_def _reset(self):**part, using /gazebo/spawn_sdf_model

```
rospy.wait_for_service('gazebo/spawn_model')
        try:
            goal_urdf = open("/home/lq/gym-gazebo/gym_gazebo/envs/assets/models/wood_cube_10cm/model.sdf","r").read()
            req = SpawnModel
            req.model_name = 'wood_cube_10cm'#the same with urdf name
            req.model_xml = goal_urdf
            req.Pose.position.x = random.uniform(1,200)
            req.Pose.position.y = random.uniform(1,200)
            self.goal.publish(req) 
        except rospy.ServiceException, e:
            print ("/gazebo/spawn_model service call failed")
```
I have import geometry msg Pose and gazebo msg SpawnModel at the very beginning and added **self.goal = rospy.Publisher('/gazebo/spawn_sdf_model',SpawnModel)** in **def __init__(self):**.
But when I python the edited _gazebo_circuit2_turtlebot_lidar.py_, the cube didn't appear.
Please help me check my code and point out how to fix it!
Many thanks!!!
Regards,
Nanxun Duo

vmayoral, 2017-12-12:
@NanxunDuo can you provide a pointer to your code and commits?

@ahcorde can you look into this?

NanxunDuo, 2017-12-12:
@vmayoral @ahcorde Thank you for your attention!
This is my commits [https://github.com/NanxunDuo/trying-gym-gazebo/blob/master/gazebo_circuit2d_turtlebot_lidar.py](url)
I only add code for publishing /gazebo/spawn_sdf_model  under _self._reset_ part.
And I wonder if I want to add extra model in gym-gazbeo environment, the _gazebo_circuit2_turtlebot_lidar.py_ is not the only file should be edited, should I also edit corresponding launch file like _GazeboCircuit2TurtlebotLidar_v0.launch_ or world file _circuit2.world_?

NanxunDuo, 2017-12-21:
I did it by /gazebo/spwan_model service!

closed 2017-12-21 by NanxunDuo