erlerobot/gym-gazebo #68: rqt_robot_dashboard error

samisaboss, 2017-12-13:
Hi, I try to install this again. 
In Ubuntu 16.04, I tried to follow all the step. It looks like all of them are fine. But when I arrive at 

```
cd gym_gazebo/envs/installation
bash setup_kinetic.bash		
```

I get a new error like the following : 

> Could not find the required component 'rqt_robot_dashboard'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
> CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
>   Could not find a package configuration file provided by
>   "rqt_robot_dashboard" with any of the following names:
> 
>     rqt_robot_dashboardConfig.cmake
>     rqt_robot_dashboard-config.cmake
> 
>   Add the installation prefix of "rqt_robot_dashboard" to CMAKE_PREFIX_PATH
>   or set "rqt_robot_dashboard_DIR" to a directory containing one of the above
>   files.  If "rqt_robot_dashboard" provides a separate development package or
>   SDK, be sure it has been installed.

Any ideas how to fix it?

vmayoral, 2017-12-14:
try removing it from https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/installation/gazebo.repos. If necessary later, install it manually.

TrinhTUHH, 2018-04-26:
This may help:
touch ~/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/kobuki_desktop/kobuki_dashboard/CATKIN_IGNORE
cd gym_gazebo/envs/installation
bash setup_kinetic.bash

nzlz, 2019-01-15:
Seems like this issue was solved, feel free to try the new installation instructions if not. Closing.

closed 2019-01-15 by nzlz