erlerobot/gym-gazebo #69: ImportError: No module named '_tf2' in Python3

zishanahmed08, 2017-12-13:
I need to import **import tf.transformations** as tft in gazebo_circuit2c_turtlebot_camera_nn.py environment.It works fine with python2. But when i shift to python3 , I get **ImportError: No module named **'_tf2****
I checked the issues raised  [here](https://github.com/ros/geometry2/issues/259).But was not able to solve it.Do you guys face the same issue on Python3?

PS 1- I took the necessary precautions related to rospy_pkg and catkin_pkg when i shifted from python 2 to 3. Infact i reinstalled everything from scratch
PS 2-The site answers.ros is down since 24 hrs ,So I am unable to get answers there

vmayoral, 2017-12-14:
Why did you re-opened this issue?

zishanahmed08, 2017-12-14:
@vmayoral -Hi victor , I closed it by mistake therefore i re-opened the issue as it still exists.Could you let me know the changes/approach to make the code compatible with python3. For example if I need to use f.transformations or bridge.imgmsg_to_cv2 in my project?

vmayoral, 2017-12-14:
@zishanahmed08 haven't tested this. You probably are the first one to do so. Can't help you right now with it as we have our hands full. If you manage to solve the issue, please report.

zishanahmed08, 2017-12-14:
@vmayoral- Ok. Thanks for the info and will report it when i can solve it.Could you please answer to my below questions?.Please excuse if the questions are pretty basic as I am new to ROS.

1)The installation instructions mention 3.5.2 as the Python version.So gym-gazebo should be python3 compatible? If I am not wrong, ros is originally not python3 compatible. So what changes are to be made to make it python3 compatible?

2)The bash setup_kinetic.bash imports a lot of packages and builds from source.For example-**gazebo_ros-packages**.By these packages are also available with **apt-get install **ros-kinetic-gazebo-ros**** .Then why are these packages built from source?Is it for python3 compatibility?If I import some new packages, Is there a flag which i can set before building to make it python3 compatible?

By basic intention is to make **gym-gazebo and open ai baselines** work together.Hence i need the gym-gazebo in python3( same as baselines).Thanks a lot for your patience

vmayoral, 2017-12-14:
> 1)The installation instructions mention 3.5.2 as the Python version.So gym-gazebo should be python3 compatible? If I am not wrong, ros is originally not python3 compatible. So what changes are to be made to make it python3 compatible?

That's not true. Last versions of ROS are Python 3 compatible.

> 2)The bash setup_kinetic.bash imports a lot of packages and builds from source.For example-gazebo_ros-packages.By these packages are also available with apt-get install ros-kinetic-gazebo-ros .Then [...]

You are mixing instructions from different sections. Focus on https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#ubuntu-1604, others are deprecated. If you find something wrong, please contribute and document it.

> By basic intention is to make gym-gazebo and open ai baselines work together.Hence i need the gym-gazebo in python3( same as baselines).Thanks a lot for your patience

As i said. It's possible. We're working with that setup.

zishanahmed08, 2017-12-14:
1) Cool.So i need not make any changes to shift from python 2 to 3 and gym-gazebo should work out of box.
@pmwenzel

2)I confused you.gazebo-ros-package was just an example.I was referring to the catkin_ws created when we run setup_kinetic.bash . I just wanted to know if we need to set sort of python-version flag in any of the CMake files there. But as you answered ROS supports python3 and no changes are necessary

3)Are u reffering to the baselines imports in GazeboModularScara4DOF-v3?

zishanahmed08, 2017-12-30:
The issue was with the compilation of catkin ws with python3. It was defaulting to python2. To avoid this I used a conda environment with default version of python3.

closed 2017-12-30 by zishanahmed08

vmayoral, 2017-12-30:
Thanks for reporting @zishanahmed08.