erlerobot/gym-gazebo #70: What are the basics to learn in able to run the sample such as turtlebot with dqn?

ylaya3, 2017-12-14:
Im a begineer, we have requirement in RL such as any simulation that a robot can learn, our professor did not teach us what software to be used, and now were doomed......i murder now 2 pc just to learn Ubuntu 14.04 and 16.04, i follow instructions in gym, ros-gym, open-ai, gym-gazebo, but still i cant run any samples, what are the things i need to learn first?im frustrated, i cant understand some steps, im a beginner in linux, please help me?help me?any step by step instructions?i'l be happy starting from a fresh formated 14.04 linux or 16.04 linux....its been 3 weeks i still cant run the sample in gym-gazebo such as tutlebot with dqn.....help me please

vmayoral, 2017-12-14:
@ylaya3 as pointed out in the README, this code requires relevant familiarity with these tools. I don't think you should start here. Please look for basic material somewhere else (ROS tutorials and Gazebo tutorials are an excellent place from where to start).

closed 2017-12-14 by vmayoral

ylaya3, 2017-12-14:
i see, thanks.....hmmm....theres no shortcut for everything....huhuhuhuhuhuhuhu.....il just accept my faith, i'l accept the consequences to be failed to this subject...