erlerobot/gym-gazebo #71: [rospack] Error: package 'hokuyo' not found

454513340, 2017-12-20:
hello，everyone，when I run an example which is called circuit2_turtlebot_lidar_qlearn.py,something wrong has happen 
as follow 
[ INFO] [1513774841.948985587, 0.110000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[rospack] Error: package 'hokuyo' not found
[librospack]: error while executing command
[FATAL] [1513774842.847947416, 12.540000000]: Package[hokuyo] does not have a path
[ INFO] [1513774843.008885122, 0.640000000]: Physics dynamic reconfigure ready.
[spawn_turtlebot_model-2] process has finished cleanly
log file: /home/exbot/.ros/log/b175c474-e585-11e7-afeb-000c29b043e8/spawn_turtlebot_model-2*.log

as far as I know ,I install the hokoyu_node package ,but what is hokuyo package?

454513340, 2017-12-22:
I have solved this problem, it may be that I have ever built a hokuyo package, and later deleted it, the system may have his records, the search package may not find that exists, so there is a problem, and then I  set up a package just fine. But new problems come out, this should be loaded kobuki driver, but how to load the DiffDriver driver?

454513340, 2017-12-22:
[image]

454513340, 2017-12-22:
I used ubuntu14.04 and ros-indigo

closed 2017-12-26 by 454513340

TS-YinJie, 2018-11-02:
I met the same error. Did you solve this error? Please teach me how to deal with it.

huangjiancong1, 2018-12-17:
I too am having this problem. 
 
I cannot see where in the URDFs hokuyo is being loaded. Only kinect calls are made. 
Any help is appreciated.

nzlz, 2018-12-17:
Hi @huangjiancong1 ,
Make sure you execute `gym-gazebo/gym_gazebo/envs/installation/turtlebot_setup.bash` before the algorithm execution.
Also make sure you have the following sources in your /.bashrc file:

- `source /opt/ros/melodic/setup.bash` (or indigo, kinetic..) <- you might be missing this one.
- `source /home/gym/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/setup.bash`

huangjiancong1, 2018-12-18:
> Hi @huangjiancong1 ,
> Make sure you execute `gym-gazebo/gym_gazebo/envs/installation/turtlebot_setup.bash` before the algorithm execution.
> Also make sure you have the following sources in your [...]

Thx @nzlz ,
But I already have some package in my ROS workstation, so I can't finish the `$ bash setup_kinetic.bash` under `/installation`, there are any method to due with this problem?

nzlz, 2018-12-18:
Take a look at [setup_kinetic.bash](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/installation/setup_kinetic.bash) and try to replicate it step by step. I recommend you use melodic.
You can have as many ros workspaces as you want, if that is what you meant.