erlerobot/gym-gazebo #72: OSError: [Errno 20] Not a directory

dangkh, 2017-12-21:
When i tried to run example python circuit_turtlebot_lidar_qlearn, i received an error. Anyone can give a help. Thanks a lot

started roslaunch server http://55cb0d51b918:33641/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [924]
ROS_MASTER_URI=http://55cb0d51b918:11311/

setting /run_id to 1a5b976a-e609-11e7-af04-0242ac110002
Gazebo launched!
... logging to /root/.ros/log/1a5b976a-e609-11e7-af04-0242ac110002/roslaunch-55cb0d51b918-947.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

process[rosout-1]: started with pid [954]
started core service [/rosout]
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 662, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 264, in _param_tag
    value = self.param_value(verbose, name, ptype, *vals)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py", line 489, in param_value
    p = subprocess.Popen(shlex.split(command), stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 711, in _init_
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 20] Not a directory

vmayoral, 2017-12-30:
Something in your setup doesn't seem right. Refer to the installation instructions and redo please. Report again if you fail at the same step.

closed 2017-12-30 by vmayoral

Dangkh109, 2017-12-30:
@vmayoral Thanks for your reply, this issue somehow appears only in newest commit, i checked out into another branch and it run again.