erlerobot/gym-gazebo #73: How I expand the observation_space in GazeboCircuit2dTurtlebotLidarEnv

NanxunDuo, 2017-12-21:
Hi there,
    I am trying to expand the observation_space in GazeboCircuit2dTurtlebotLidarEnv.
    For now the observation in GazeboCircuit2dTurtlebotLidarEnv only includes discretized_ranges, and I wonder if I can add another sensor message in observation.
   I noticed that action_space can be set as spaces.Discrete or Box or other kind of spaces.
   But I read openai gym docs and didn't find any explanation of how to set observation_space.
   Any ideas of setting observation_space?
   Thanks for reading this issue and I will be very appreciate for your valuable suggestion!!

zengjie617789, 2018-03-26:
had you find a solutions to it?

nzlz, 2019-01-15:
@NanxunDuo if you are still interested, make sure to open a new **Advice request** issue to facilitate issue-tracking, or post your thought directly at https://github.com/openai/gym/issues.

closed 2019-01-15 by nzlz