erlerobot/gym-gazebo #75: No module named 'baselines.agent'

jerpint, 2017-12-30:
When trying to run the modular_scara example file
`python3 modular_scara_3dof_random.py`

I get the following error:

`ImportError: No module named 'baselines.agent'`

The traceback leads me back to the file `gazebo_modular_scara_3dof.py` which has the following hint on line 27 :

```
# from custom baselines repository
from baselines.agent.utility.general_utils import forward_kinematics, get_ee_points, rotation_from_matrix, \
    get_rotation_matrix,quaternion_from_matrix # For getting points and velocities.
```

Where is the baselines.agent library located??

vmayoral, 2017-12-30:
@jerpint, we're still working on the `GazeboModularScara` environments. They aren't available just yet. You will get tons of dependency issues like that one until we finish it up.

closed 2017-12-30 by vmayoral

jerpint, 2017-12-30:
@vmayoral ok great, good to know, is there an expected ETA on that?? Thanks for the great work :)

vmayoral, 2017-12-30:
Thanks for your interest @jerpint, sorry, don't want to over promise so no ETA for now. I can tell you though that we're working hard on this and a few other new environments.

For now I made a quick modification in the README to reflect the WIP environments and the ones that are community supported.

jerpint, 2017-12-30:
@vmayoral seeing as this is still WIP, how much would you like new users like myself to report bugs/issues?

vmayoral, 2017-12-31:
Hello @jerpint, regarding the overall infrastructure and the "community-environments". Feel free to report as much as you'd like.