erlerobot/gym-gazebo #81: Requirement of Gazebo8

borongyuan, 2018-01-12:
Hi,
I've got some problem with Gazebo8 and ros_control, but there's no problem with Gazebo7.
So I'd like to know, is Gazebo8 necessary for gym-gazebo? Can I switch to Gazebo7?
Thanks

vmayoral, 2018-01-12:
>  Can I switch to Gazebo7?

Yes but you'll probably need to fix some additional dependency issues.

jacknlliu, 2018-01-12:
@vmayoral I have a question that why this repository shipped with Gazebo8 not gazebo7 which works with ROS kinetic as default.

vmayoral, 2018-01-12:
@jacknlliu, we just wanted to exploit some of the functionalities of Gazebo 8.

zishanahmed08, 2018-01-12:
@jacknlliu -follow the below installation instructions for gazebo8 compatibility

sudo apt-get remove ros-*
#to remove existing ros

sudo apt-get install ros-kinetic-desktop
#dont install ros-kinetic-desktop-**full** since it installs gazebo7

sudo rosdep init
rosdep update
sudo apt remove '.*gazebo.*' '.*sdformat.*' '.*ignition-.*'
#removes existing gazebo7 installation files if any

sudo apt-get install gazebo8
sudo apt-get install ros-kinetic-gazebo8-ros-pkgs ros-kinetic-gazebo8-ros-control
#install gazebo8 specific ros packages

vmayoral, 2018-01-12:
Thanks for providing detailed instructions @zishanahmed08, I've made a slight syntax change. +1!

borongyuan, 2018-01-14:
Thanks. It looks like they have just made some mistakes, using wrong dependency for the latest ros-kinetic-gazebo8-ros.
https://bitbucket.org/osrf/gazebo/issues/2399/dependency-problem

borongyuan, 2018-02-13:
I had solved my issue. It was related to rqt plugin, not Gazebo. So closing...

closed 2018-02-13 by borongyuan