erlerobot/gym-gazebo #82: Tried to reset environment which is not done

365220121, 2018-01-12:
> `EP 203 - 1001 timesteps - last100 Steps : 146 - Cumulated R: 2214.55   Eps=0.36     Time: 0:53:49
 Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_dqn.py", line 334, in <module>
    observation = env.reset()
  File "/home/yang/gym/gym/core.py", line 104, in reset
    return self._reset()
  File "/home/yang/gym/gym/wrappers/monitoring.py", line 38, in _reset
    self._before_reset()
  File "/home/yang/gym/gym/wrappers/monitoring.py", line 189, in _before_reset
    self.stats_recorder.before_reset()
  File "/home/yang/gym/gym/monitoring/stats_recorder.py", line 68, in before_reset
    raise error.Error("Tried to reset environment which is not done. While the monitor is active for {}, you cannot call reset() unless the episode is over.".format(self.env_id))
gym.error.Error: Tried to reset environment which is not done. While the monitor is active for GazeboCircuit2TurtlebotLidarNn-v0, you cannot call reset() unless the episode is over.`

when i python 'circuit2_turtlebot_lidar_dqn.py',i met wiht this error.I tried to debug it by google and  github for a while.But i am failed！ So i am confused.Could you give me advice to debug it well.
[circuit2_turtlebot_lidar_dqn .txt](https://github.com/erlerobot/gym-gazebo/files/1625405/circuit2_turtlebot_lidar_dqn.txt)

YantianZha, 2018-04-26:
I'm having the same problem.

castleguarders, 2018-05-30:
Same issue, spend many hours trying out different combinations of gym version, all the way  from .10.5 to .7.4.

yangningbocn, 2018-05-31:
max_episode_steps=1000

in

gym-gazebo/gym_gazebo/init.py

in the registry of the env I am using. But I think this fix is cleaner.

others told me to do this. And i fix it too.Hope to help you.
If not ,email me ,I will try me best to help you .

castleguarders, 2018-05-31:
I saw this mentioned elsewhere, but it was already set correctly for the Nn-v0 entry, i didn't realize that Lidar-v0 register should also have had the entry. For future reference, the file in my environment is named__init__.py 

Your note made me take a closer look, with the added entry, it worked! Thank you.

register(
    id='GazeboCircuit2TurtlebotLidar-v0',
    entry_point='gym_gazebo.envs.turtlebot:GazeboCircuit2TurtlebotLidarEnv',
    # More arguments here
)
register(
    id='GazeboCircuit2TurtlebotLidarNn-v0',
    entry_point='gym_gazebo.envs.turtlebot:GazeboCircuit2TurtlebotLidarNnEnv',
    max_episode_steps=1000,
    # More arguments here
)

closed 2018-05-31 by yangningbocn