erlerobot/gym-gazebo #83: Asynchronous cognitive architecture

vmayoral, 2018-01-12:
From https://link.springer.com/chapter/10.1007/978-3-319-63940-6_28:

> Therefore, researchers aim to easily apply the asynchronous cognitive architecture to solve RL problems. The toolkit, called Gym-Gazebo [4 ], is used to develop RL algorithms. It extends Gym [3] using a robot operation system (ROS) [8 ] and Gazebo [6 ] and can use robots. Gym provides environments in video games, such as Pong and Go, and a toolkit for developing and comparing RL algorithms. However, Gym-Gazebo has a problem in applying the asynchronous cognitive architecture because of Gym. Gym assumes synchrony, and all actions or policies are executed in synchrony. Gym-Gazebo is scalable to a 3D simulator and robots in the real world, but does not solve the difference between synchrony and asynchrony. Moreover, it cannot apply the asynchronous cognitive architecture.

I believe this is good criticism. Environments like `GazeboModularScara3DOF-v2` has been programmed synchronously however newer environments such as `GazeboModularScara3DOF-v3`are actually working in an asynchronous way (learning technique is sampling the environment when it deems necessary).

I think this topic is worth discussing. @rkojcev, @noraetxezarreta and @ahcorde, feel free to provide input here.

I've additionally reached out the contact information of the paper inviting them to contribute and share their opinion.

jacknlliu, 2018-01-12:
Does this environment like GazeboModularScara3DOF-v2 work in this repository now? @vmayoral

nzlz, 2019-01-15:
@jacknlliu if you are still interested in this matter, please open a new issue to facilitate the issue-tracking. Closing.

closed 2019-01-15 by nzlz