erlerobot/gym-gazebo #84: OS X support

vmayoral, 2018-01-12:
This issue will discuss an attempt to make `gym-gazebo` work in OS X.

vmayoral, 2018-01-12:
The process will be executed in OS X `10.13.1`:

- Install Gazebo 8
- Install ROS Kinetic: http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source

Particular details:
```bash
wstool init -j1 src kinetic-ros_comm-wet.rosinstall
./src/catkin/bin/catkin_make_isolated -DPYTHON_VERSION=3.6 --install -DCMAKE_BUILD_TYPE=Release
```

Fixes( packages that I manually introduced in the workspace, installed by hand or ignored):
- https://github.com/ros/console_bridge
- `brew install lz4`
- CATKIN_IGNORE `ros_comm/rosbag_storage`
- https://discussions.apple.com/thread/7707202

After that, it worked:
```
$ roscore 
... logging to /Users/victor/.ros/log/16b20602-f79e-11e7-a337-a8667f3838a8/roslaunch-Victors-MacBook.local-44996.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt

started roslaunch server http://Victors-MacBook.local:51911/
ros_comm version 1.12.12

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES

auto-starting new master
process[master]: started with pid [45004]
ROS_MASTER_URI=http://Victors-MacBook.local:11311/

setting /run_id to 16b20602-f79e-11e7-a337-a8667f3838a8
process[rosout-1]: started with pid [45007]
started core service [/rosout]
```
- Python `3.6.0` installed from conda
- installed sophus:
```
git clone http://github.com/stonier/sophus -b indigo
cd sophus/
mkdir build; cd build; cmake ..
make
history
sudo make install
```
- Installed gym-gazebo

vmayoral, 2018-01-12:
Added the following dependencies to the ROS (Kinetic) workspace:
- https://github.com/ros/common_msgs
- https://github.com/ros-controls/control_msgs
- github.com/erlerobot/orocos_kinematics_dynamics

After doing this, we get a problem with `sip` library, currently: ` sip: 4.18-py36_0  ` from conda produces:
```
python3 modular_scara_3dof_random.py 
[2018-01-12 15:22:48,525] Making new env: GazeboModularScara3DOF-v2
Segmentation fault: 11
```

The following versions are available:
```
conda search -f sip
Fetching package metadata ...........
sip                          4.15.5                   py26_0  defaults        
                             4.15.5                   py27_0  defaults        
                             4.15.5                   py33_0  defaults        
                             4.15.5                   py34_0  defaults        
                             4.16.5                   py26_0  defaults        
                             4.16.5                   py27_0  defaults        
                             4.16.5                   py33_0  defaults        
                             4.16.5                   py34_0  defaults        
                             4.16.5                   py35_0  defaults        
                             4.16.9                   py27_0  defaults        
                             4.16.9                   py34_0  defaults        
                             4.16.9                   py35_0  defaults        
                             4.18                     py27_0  defaults        
                             4.18                     py34_0  defaults        
                             4.18                     py35_0  defaults        
                             4.18                     py36_0  defaults        
                             4.18.1           py27h6300f65_2  defaults        
                             4.18.1           py35h79e1f92_2  defaults        
                             4.18.1           py36h2824476_2  defaults      
```

vmayoral, 2018-01-12:
With `sip=4.18.1=py36h2824476_2`:
```
$ python
Python 3.6.0 |Anaconda custom (x86_64)| (default, Dec 23 2016, 13:19:00) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyKDL as kdl
Segmentation fault: 11
```

vmayoral, 2018-01-12:
Tried compiling and installing sip from sources, particularly the https://sourceforge.net/projects/pyqt/files/sip/sip-4.17/ version.

When recompiling `python_orocos_kdl`:
```
...
Scanning dependencies of target python_module_PyKDL
[ 10%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart0.o
[ 20%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.o
[ 40%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart3.o
[ 40%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.o
In file included from /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipPyKDLpart2.cpp:7:
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:10:10: fatal error: 
      'sip.h' file not found
#include <sip.h>
         ^~~~~~~
In file included from /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipPyKDLpart0.cpp:7:
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:10:10: fatal error: 
      'sip.h' file not found
#include <sip.h>
         ^~~~~~~
In file included from /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipPyKDLpart3.cpp:7:
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:10:10: fatal error: 
      'sip.h' file not found
#include <sip.h>
         ^~~~~~~
In file included from /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipPyKDLpart1.cpp:7:
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:10:10: fatal error: 
      'sip.h' file not found
#include <sip.h>
         ^~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart3.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
1 error generated.
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.o] Error 1
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart0.o] Error 1
1 error generated.
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.o] Error 1
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'python_orocos_kdl': 
  Command '['/Users/victor/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl && /Users/victor/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.
victor at Victors-MacBook in ~/ros_catkin_ws
$ 
victor at Victors-MacBook in ~/ros_catkin_ws
$ 
victor at Victors-MacBook in ~/ros_catkin_ws
$ PYTHONPATH./src/catkin/bin/catkin_make_isolated -DPYTHON_VERSION=3.6 --install -DCMAKE_BUILD_TYPE=Release --pkg python_orocos_kdl
victor at Victors-MacBook in ~/ros_catkin_ws
$ 
victor at Victors-MacBook in ~/ros_catkin_ws
$ 
victor at Victors-MacBook in ~/ros_catkin_ws
$ ./src/catkin/bin/catkin_make_isolated -DPYTHON_VERSION=3.6 --install -DCMAKE_BUILD_TYPE=Release --pkg python_orocos_kdl
Base path: /Users/victor/ros_catkin_ws
Source space: /Users/victor/ros_catkin_ws/src
Build space: /Users/victor/ros_catkin_ws/build_isolated
Devel space: /Users/victor/ros_catkin_ws/devel_isolated
Install space: /Users/victor/ros_catkin_ws/install_isolated
Additional CMake Arguments: -DPYTHON_VERSION=3.6 -DCMAKE_BUILD_TYPE=Release
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 63 packages in topological order:
~~  - catkin
~~  - genmsg
[... 530 lines cut ...]
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:695:51: note: 
      expanded from macro 'sipIsOwnedByPython'
#define sipIsOwnedByPython          sipAPI_PyKDL->api_is_owned_by_python
                                    ~~~~~~~~~~~~  ^
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipPyKDLpart2.cpp:1091:63: error: 
      no member named 'api_is_derived_class' in '_sipAPIDef'
        release_ChainIkSolverVel_pinv(sipGetAddress(sipSelf), sipIsDerivedClass(sipSel...
                                                              ^~~~~~~~~~~~~~~~~
/Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl/install/PyKDL/sipAPIPyKDL.h:696:51: note: 
      expanded from macro 'sipIsDerivedClass'
#define sipIsDerivedClass           sipAPI_PyKDL->api_is_derived_class
                                    ~~~~~~~~~~~~  ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.o] Error 1
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'python_orocos_kdl': 
  Command '['/Users/victor/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws/build_isolated/python_orocos_kdl && /Users/victor/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.
```

vmayoral, 2018-01-12:
Managed to fix this issue installing `sip-4.19.6` by hand from https://sourceforge.net/projects/pyqt/files/sip/.

Next up:
```
python modular_scara_3dof_random.py 
[2018-01-12 15:57:19,158] Making new env: GazeboModularScara3DOF-v2
Traceback (most recent call last):
  File "modular_scara_3dof_random.py", line 3, in <module>
    env = gym.make('GazeboModularScara3DOF-v2')
  File "/Users/victor/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 161, in make
    return registry.make(id)
  File "/Users/victor/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/Users/victor/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/Users/victor/gym-gazebo/gym_gazebo/envs/modular_scara/gazebo_modular_scara_3dof_v2.py", line 58, in __init__
    gazebo_env.GazeboEnv.__init__(self, "ModularScara3_v0.launch")
  File "/Users/victor/gym-gazebo/gym_gazebo/envs/gazebo_env.py", line 20, in __init__
    port = os.environ["ROS_PORT_SIM"]
  File "/Users/victor/anaconda3/lib/python3.6/os.py", line 669, in __getitem__
    raise KeyError(key) from None
KeyError: 'ROS_PORT_SIM'
```

vmayoral, 2018-01-12:
```
 export ROS_PORT_SIM="11311"
export GYM_GAZEBO_ENV_SCARA3="/Users/victor/gym-gazebo/gym_gazebo/envs/assets/worlds/scara_basic.world" 
```

ahcorde, 2018-01-12:
This file contains my indigo ROS src

[gazebo_osx.txt](https://github.com/erlerobot/gym-gazebo/files/1626507/gazebo_osx.txt)

vmayoral, 2018-01-12:
Missing `https://github.com/ros-simulation/gazebo_ros_pkgs` dependencies:
- https://github.com/ros/geometry
- https://github.com/ros-simulation/gazebo_ros_pkgs
- https://github.com/ros/geometry2

vmayoral, 2018-01-12:
Getting the following errors:

```
==> make -j4 -l4 in '/Users/victor/ros_catkin_ws2/build_isolated/rosbag_storage'
Scanning dependencies of target rosbag_storage
[  8%] Building CXX object CMakeFiles/rosbag_storage.dir/src/bag.cpp.o
[ 16%] Building CXX object CMakeFiles/rosbag_storage.dir/src/bz2_stream.cpp.o
/Users/victor/ros_catkin_ws2/src/ros_comm/ros_comm/tools/rosbag_storage/src/bz2_stream.cpp:130:13: error: 
      use of undeclared identifier 'logError'
            logError("unused data already available");
            ^
1 error generated.
make[2]: *** [CMakeFiles/rosbag_storage.dir/src/bz2_stream.cpp.o] Error 1
make[1]: *** [CMakeFiles/rosbag_storage.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
<== Failed to process package 'rosbag_storage': 
  Command '['/Users/victor/ros_catkin_ws2/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws2/build_isolated/rosbag_storage && /Users/victor/ros_catkin_ws2/install_isolated/env.sh make -j4 -l4

Command failed, exiting.
```

vmayoral, 2018-01-13:
Ended up installing `console_bridge` from brew.

More dependencies:
- https://github.com/ros-controls/control_toolbox
- https://github.com/ros/dynamic_reconfigure
- https://github.com/ros-controls/realtime_tools
- https://github.com/ros/actionlib
- https://github.com/ros/pluginlib
- https://github.com/ros/class_loader
- `brew install urdfdom`
- https://github.com/ros/urdf

vmayoral, 2018-01-13:
Stucked compiling `gazebo_ros`:
```
==> Processing catkin package: 'gazebo_ros'
==> Building with env: '/Users/victor/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/Users/victor/ros_catkin_ws/build_isolated/gazebo_ros'
==> make -j4 -l4 in '/Users/victor/ros_catkin_ws/build_isolated/gazebo_ros'
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target gazebo_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target tf_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target gazebo_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target actionlib_generate_messages_lisp
[ 20%] Built target gazebo_ros_gencfg
[ 20%] Built target std_msgs_generate_messages_cpp
[ 20%] Built target actionlib_generate_messages_cpp
[ 20%] Built target sensor_msgs_generate_messages_nodejs
[ 20%] Built target actionlib_generate_messages_nodejs
[ 20%] Built target sensor_msgs_generate_messages_lisp
[ 20%] Built target std_srvs_generate_messages_nodejs
[ 20%] Built target tf_generate_messages_py
[ 20%] Built target sensor_msgs_generate_messages_eus
[ 20%] Built target dynamic_reconfigure_generate_messages_py
[ 20%] Built target geometry_msgs_generate_messages_eus
[ 20%] Built target actionlib_generate_messages_py
[ 20%] Built target tf_generate_messages_nodejs
[ 20%] Built target trajectory_msgs_generate_messages_nodejs
[ 20%] Built target std_msgs_generate_messages_nodejs
[ 20%] Built target rosgraph_msgs_generate_messages_eus
[ 20%] Built target tf_generate_messages_lisp
[ 20%] Built target dynamic_reconfigure_generate_messages_lisp
[ 20%] Built target std_srvs_generate_messages_lisp
[ 20%] Built target gazebo_msgs_generate_messages_cpp
[ 20%] Built target roscpp_generate_messages_py
[ 20%] Built target std_srvs_generate_messages_eus
[ 20%] Built target tf_generate_messages_cpp
[ 20%] Built target std_srvs_generate_messages_py
[ 20%] Built target std_srvs_generate_messages_cpp
[ 20%] Built target gazebo_msgs_generate_messages_py
[ 20%] Built target std_msgs_generate_messages_eus
[ 20%] Built target geometry_msgs_generate_messages_nodejs
[ 20%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 20%] Built target roscpp_generate_messages_lisp
[ 20%] Built target actionlib_generate_messages_eus
[ 20%] Built target roscpp_generate_messages_cpp
make[2]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/Accelerate.framework', needed by `/Users/victor/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_api_plugin.dylib'.  Stop.
make[1]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
<== Failed to process package 'gazebo_ros': 
  Command '['/Users/victor/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws/build_isolated/gazebo_ros && /Users/victor/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.

```

vmayoral, 2018-01-13:
Tried with `lunar-devel` and `indigo-devel`, still failing:

```
 ./src/catkin/bin/catkin_make_isolated -DPYTHON_VERSION=3.6 --install -DCMAKE_BUILD_TYPE=Release --pkg gazebo_ros
Base path: /Users/victor/ros_catkin_ws
Source space: /Users/victor/ros_catkin_ws/src
Build space: /Users/victor/ros_catkin_ws/build_isolated
Devel space: /Users/victor/ros_catkin_ws/devel_isolated
Install space: /Users/victor/ros_catkin_ws/install_isolated
Additional CMake Arguments: -DPYTHON_VERSION=3.6 -DCMAKE_BUILD_TYPE=Release
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 120 packages in topological order:
~~  - catkin
~~  - genmsg
~~  - gencpp
~~  - geneus
~~  - genlisp
~~  - gennodejs
~~  - genpy
~~  - cmake_modules
~~  - class_loader
~~  - common_msgs
~~  - cpp_common
~~  - gazebo_dev
~~  - gazebo_ros_pkgs
~~  - geometry
~~  - geometry2
~~  - geometry_experimental
~~  - message_generation
~~  - message_runtime
~~  - mk
~~  - orocos_kdl (plain cmake)
~~  - orocos_kinematics_dynamics
~~  - python_orocos_kdl (plain cmake)
~~  - ros
~~  - ros_comm
~~  - ros_control
~~  - rosbash
~~  - rosboost_cfg
~~  - rosbuild
~~  - rosclean
~~  - roscpp_traits
~~  - roscreate
~~  - rosgraph
~~  - roslang
~~  - roslisp
~~  - rosmake
~~  - rosmaster
~~  - rospack
~~  - roslib
~~  - rosparam
~~  - rospy
~~  - rosservice
~~  - rostime
~~  - roscpp_serialization
~~  - roslaunch
~~  - rosunit
~~  - angles
~~  - rosconsole
~~  - pluginlib
~~  - roslz4
~~  - rosbag_storage
~~  - rostest
~~  - std_msgs
~~  - actionlib_msgs
~~  - controller_manager_msgs
~~  - diagnostic_msgs
~~  - geometry_msgs
~~  - eigen_conversions
~~  - kdl_conversions
[... 557 lines cut ...]
[ 20%] Built target roscpp_generate_messages_cpp
Scanning dependencies of target trajectory_msgs_generate_messages_py
[ 20%] Built target trajectory_msgs_generate_messages_py
Scanning dependencies of target gazebo_msgs_generate_messages_nodejs
[ 20%] Built target gazebo_msgs_generate_messages_nodejs
Scanning dependencies of target gazebo_msgs_generate_messages_lisp
[ 20%] Built target gazebo_msgs_generate_messages_lisp
Scanning dependencies of target trajectory_msgs_generate_messages_cpp
[ 20%] Built target trajectory_msgs_generate_messages_cpp
Scanning dependencies of target gazebo_ros_api_plugin
make[2]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/Accelerate.framework', needed by `/Users/victor/ros_catkin_ws/devel_isolated/gazebo_ros/lib/libgazebo_ros_api_plugin.dylib'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 40%] Building CXX object CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o
make[1]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
<== Failed to process package 'gazebo_ros': 
  Command '['/Users/victor/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws/build_isolated/gazebo_ros && /Users/victor/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.

```

vmayoral, 2018-01-13:
Got tons of issues like:
```
Your PYTHONPATH points to a site-packages dir for Python 2.x but you are running Python 3.x!
     PYTHONPATH is currently: "/Users/victor/ros_catkin_ws2/install_isolated/lib/python3.6/site-packages:/Users/victor/ros_catkin_ws2/build_isolated/catkin/lib/python3.6/site-packages:"
     You should `unset PYTHONPATH` to fix this.
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/Users/victor/ros_catkin_ws2/build_isolated/catkin/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:149 (include)

make: *** [install] Error 1
<== Failed to process package 'catkin': 
  Command '['make', 'install']' returned non-zero exit status 2.

```

Apparently, an issue between `brew` and `conda`.

vmayoral, 2018-01-13:
Still getting 

```
Your PYTHONPATH points to a site-packages dir for Python 2.x but you are running Python 3.x!
```

The following fixed it:
```
mv /usr/local/lib/python2.7/site-packages/sitecustomize.py /usr/local/lib/python2.7/site-packages/sitecustomize.py.aux
```

vmayoral, 2018-01-13:
Follow up issues with ROS packages:
- https://github.com/mikepurvis/ros-install-osx/issues/91
- https://github.com/ros/rospack/commit/6fcee1b0d2b1c225c39a5820981166b99f0d8750

vmayoral, 2018-01-13:
Again stucked at:
```
==> Processing catkin package: 'gazebo_ros'
==> Building with env: '/Users/victor/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/Users/victor/ros_catkin_ws/build_isolated/gazebo_ros'
==> make -j4 -l4 in '/Users/victor/ros_catkin_ws/build_isolated/gazebo_ros'
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target tf_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target gazebo_msgs_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[ 20%] Built target gazebo_ros_gencfg
[ 20%] Built target roscpp_generate_messages_eus
[ 20%] Built target actionlib_generate_messages_lisp
[ 20%] Built target tf2_msgs_generate_messages_lisp
[ 20%] Built target tf2_msgs_generate_messages_eus
[ 20%] Built target tf2_msgs_generate_messages_cpp
[ 20%] Built target actionlib_msgs_generate_messages_nodejs
[ 20%] Built target actionlib_msgs_generate_messages_lisp
[ 20%] Built target std_srvs_generate_messages_nodejs
[ 20%] Built target actionlib_msgs_generate_messages_cpp
[ 20%] Built target tf_generate_messages_nodejs
[ 20%] Built target trajectory_msgs_generate_messages_nodejs
[ 20%] Built target std_msgs_generate_messages_nodejs
[ 20%] Built target rosgraph_msgs_generate_messages_eus
[ 20%] Built target tf_generate_messages_lisp
[ 20%] Built target dynamic_reconfigure_generate_messages_lisp
[ 20%] Built target std_srvs_generate_messages_lisp
[ 20%] Built target gazebo_msgs_generate_messages_cpp
[ 20%] Built target roscpp_generate_messages_py
[ 20%] Built target std_srvs_generate_messages_eus
[ 20%] Built target tf_generate_messages_cpp
[ 20%] Built target std_srvs_generate_messages_py
[ 20%] Built target gazebo_msgs_generate_messages_py
[ 20%] Built target std_srvs_generate_messages_cpp
[ 20%] Built target std_msgs_generate_messages_eus
[ 20%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 20%] Built target roscpp_generate_messages_lisp
[ 20%] Built target geometry_msgs_generate_messages_nodejs
[ 20%] Built target geometry_msgs_generate_messages_cpp
[ 20%] Built target actionlib_msgs_generate_messages_eus
[ 20%] Built target trajectory_msgs_generate_messages_lisp
[ 20%] Built target std_msgs_generate_messages_py
[ 20%] Built target rosgraph_msgs_generate_messages_py
[ 20%] Built target rosgraph_msgs_generate_messages_nodejs
[ 20%] Built target std_msgs_generate_messages_lisp
[ 20%] Built target rosgraph_msgs_generate_messages_lisp
[ 20%] Built target actionlib_msgs_generate_messages_py
[ 20%] Built target sensor_msgs_generate_messages_py
[ 20%] Built target geometry_msgs_generate_messages_lisp
[ 20%] Built target roscpp_generate_messages_nodejs
[ 20%] Built target actionlib_generate_messages_eus
[ 20%] Built target sensor_msgs_generate_messages_cpp
[ 20%] Built target trajectory_msgs_generate_messages_eus
[ 20%] Built target rosgraph_msgs_generate_messages_cpp
[... 17 lines cut ...]
```

Possible reasons:
1.  boost
```
CMake Warning at /usr/local/Cellar/cmake/3.10.1/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.10.1/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/Cellar/cmake/3.10.1/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:146 (find_package)
  /Users/victor/ros_catkin_ws/install_isolated/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:2 (find_package)
  /Users/victor/ros_catkin_ws/install_isolated/share/gazebo_dev/cmake/gazebo_devConfig.cmake:197 (include)
  /Users/victor/ros_catkin_ws/install_isolated/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:4 (find_package)
```
2. paths
```
MACOSX_RPATH is not specified for the following targets:

   gazebo_ros_api_plugin
   gazebo_ros_paths_plugin

```

zishanahmed08, 2018-01-13:
Not sure if this will help you.But i faced similar issues with compiling boost related issue on Ubuntu 16.04

'For solving the boost package error related to c make refer this post https://github.com/BVLC/caffe/issues/4843'
sudo ln -s libboost_python-py35.so libboost_python3.so

vmayoral, 2018-01-13:
thanks @zishanahmed08 for the help, i believed i moved beyond that issue by reinstalling Gazebo 8 and all its corresponding dependencies. Appreciate the pointer on a Saturday though!

Here's my status:
```
[ 40%] Building CXX object CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Actor.hh:24:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Entity.hh:25:
In file included from /usr/local/Cellar/ignition-transport3/3.1.0/include/ignition/transport3/ignition/transport/Node.hh:32:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:17:2: error: This file
      was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:18:2: error: 
      incompatible with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:19:2: error: 
      regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Actor.hh:24:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Entity.hh:25:
In file included from /usr/local/Cellar/ignition-transport3/3.1.0/include/ignition/transport3/ignition/transport/Node.hh:32:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:32:
/usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:17:2: error: This file was
      generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:18:2: error: incompatible
      with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:19:2: error: regenerate this
      file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
In file included from /Users/victor/ros_catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Actor.hh:24:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Entity.hh:25:
In file included from /usr/local/Cellar/ignition-transport3/3.1.0/include/ignition/transport3/ignition/transport/Node.hh:32:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:33:
/usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:17:2: error: This file was
      generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:18:2: error: incompatible
      with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:19:2: error: regenerate
      this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
[... 69 lines cut ...]
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo8/8.2.0_1/include/gazebo-8/gazebo/physics/Entity.hh:25:
In file included from /usr/local/Cellar/ignition-transport3/3.1.0/include/ignition/transport3/ignition/transport/Node.hh:32:
[... cut ...]
make[2]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_api_plugin.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'gazebo_ros': 
  Command '['/Users/victor/ros_catkin_ws/install_isolated/env.sh', 'make', '-j1']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /Users/victor/ros_catkin_ws/build_isolated/gazebo_ros && /Users/victor/ros_catkin_ws/install_isolated/env.sh make -j1

Command failed, exiting.

```

vmayoral, 2018-01-13:
A few more bits:
```
victor at Victors-MacBook in ~/ros_catkin_ws
$ protoc --version
libprotoc 3.5.1
victor at Victors-MacBook in ~/ros_catkin_ws
$ brew info protobuf
protobuf: stable 3.5.1 (bottled), HEAD
Protocol buffers (Google's data interchange format)
https://github.com/google/protobuf/
/usr/local/Cellar/protobuf/3.1.0 (277 files, 16.4MB)
  Poured from bottle on 2016-10-30 at 13:41:18
/usr/local/Cellar/protobuf/3.3.2 (164 files, 13.9MB)
  Poured from bottle on 2017-07-09 at 19:44:10
/usr/local/Cellar/protobuf/3.5.1 (267 files, 18.0MB) *
  Poured from bottle on 2018-01-13 at 15:55:30
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/protobuf.rb
==> Dependencies
Build: autoconf ✔, automake ✘, libtool ✔
Optional: python3 ✔
==> Options
--with-python3
	Build with python3 support
--with-test
	Run build-time check
--without-python
	Build without python support
--HEAD
	Install HEAD version
==> Caveats
Editor support and examples have been installed to:
  /usr/local/opt/protobuf/share/doc/protobuf

This formula installed .pth files to Homebrew's site-packages and your
Python isn't configured to process them, so you will not be able to
import the modules this formula installed. If you plan to develop
with these modules, please run:
  mkdir -p /Users/victor/Library/Python/2.7/lib/python/site-packages
  echo 'import site; site.addsitedir("/usr/local/lib/python2.7/site-packages")' >> /Users/victor/Library/Python/2.7/lib/python/site-packages/homebrew.pth
```

vmayoral, 2018-01-13:
Tried changing protobuf versions, no luck.

Abandoning this for now.

zishanahmed08, 2018-01-14:
Hi Victor, I  had faced the similar issue of protobuf versions in Ubuntu 16.04.The solution mentioned
[here](https://github.com/BVLC/caffe/issues/5645).Refer to the comment by limTeTE  regarding conflicting versions of protobuf installed by Conda. Make sure that you do not install any application (like tensorflow) which upgrades your protobuf after installation of ROS and before building your catkin-ws. I assume that a older version of protobuf was present during ROS installation which was upgraded by conda before you built your catkin_ws.I am not a 100% sure about this.

closed 2018-01-27 by vmayoral