erlerobot/gym-gazebo #85: PPO implementation for Turtlebot envs

zishanahmed08, 2018-01-12:
This is more of a question rather than issue. I am trying to impelement PPO algorithm on the turtlebot.
I check your example implementation for [scara_PPO](https://github.com/erlerobot/gym-gazebo/blob/master/examples/modular_scara/modular_scara_3dof_ppo.py).I assume it is based on  [run_mujoco](https://github.com/openai/baselines/blob/master/baselines/ppo1/run_mujoco.py).

The following changes would be required to get PPO implementation for Turtlebot envs
1. Use [run_atari](https://github.com/openai/baselines/blob/master/baselines/ppo1/run_atari.py) since it is based on atari games with discrete action space similar to turtlebot
2. Shift to Cnn policy from MLP policy

I have the following queries when using run_atari as base
1. It uses MPI for parallel execution. Does Gym-gazebo turtlebot envs support this?
2.How do i deal with Atari and deepming wrappers used (make_atari, wrap_deepmind) ,which takes care of frame stacking etc.
3.A newer implementation ppo3 optimised for Multi gpu is available.Can it be used?

**PS**- I understand that Scara is WIP and is not supported .But it would be great if you could give your thoughts on my above questions.

vmayoral, 2018-01-12:
Hello @zishanahmed08,
> I check your example implementation for scara_PPO.I assume it is based on run_mujoco.

that's right.

> It uses MPI for parallel execution. Does Gym-gazebo turtlebot envs support this?
Latest experiments are running in an asynchronous way while using MPI ([pointer](http://ndnsim.net/2.0/parallel-simulations.html) about its integration in simulation) but not with the turtlebot envs. Review https://github.com/erlerobot/gym-gazebo/issues/83 where this is being discussed. Feel free to check those envs where we're trying to get the asynchronous approach working and try it out in the turtlebot. I don't see why it shouldn't work.

> 2.How do i deal with Atari and deepming wrappers used (make_atari, wrap_deepmind) ,which takes care of frame stacking etc.

This is a more algorithmic part and not related strictly to the environment itself. With the information provided i can't follow.

> 3.A newer implementation ppo3 optimised for Multi gpu is available.Can it be used?

I wasn't aware of any public `ppo3` version just yet but yes, I'd say once available it's worth giving it a try.

zishanahmed08, 2018-01-12:
@vmayoral Thanks Victor for the info.I will look into the asynchronous implementation in the new scara envs. 
there was a typo I meant ppo2 and not ppo3. I have updated the earlier comment.

vmayoral, 2018-01-12:
> here was a typo I meant ppo2 and not ppo3. I have updated the earlier comment.

I see! That makes sense. We're trying `ppo2` actively and getting good results with (in simulation)  but surprisingly when we transfer the behaviors to the real robots, the results are a bit disappointing. I believe the algorithm is currently being refactored by the OpenAI guys so we'll need to stay tuned.

closed 2018-01-12 by vmayoral

zishanahmed08, 2018-01-23:
@vmayoral After a comparison GazeboModularScara3DOF-v2 & GazeboModularScara3DOF-v3, 
I understand the below changes were necessary for making scara/turtlebot environment asynchronous

**Previous synchronous method** 
1. Unpause simulation
2. Publish msg to go to initial state(take action)
3. Sleep for time= slowness
3. Take Observation
4. Unpause simulation

**Current asynchronous method** 
1. Take action(reach the specified target position in 'slowness' time)
2. Take Observation

Please confirm if my understanding is right.

I also have 2 more questions as below

1.`def init_time(self, slowness =1, slowness_unit='sec', reset_jnts=True):` In V3 , Function init_time is defined but never used.What am i missing?
2.`def _reset(self):` In V3, in  reset call to `rospy.wait_for_service('/gazebo/reset_simulation')` has been removed.Why so?

PS- I am taging you, since I am not sure if the notifications are delivered to you after issue is closed

rkojcev, 2018-01-23:
@zishanahmed08 we recently added `def init_time(self, slowness =1, slowness_unit='sec', reset_jnts=True)` for some of our experiments, that is not being used in turtlebot and it will not change anything in your code. The default time is set to be in seconds
the code: `rospy.wait_for_service('/gazebo/reset_simulation')` is being removed because we do not reset the simulation anymore.

zishanahmed08, 2018-01-23:
Thanks for the explanation.
Regarding `rospy.wait_for_service('/gazebo/reset_simulation')` , I believe in scara there is no collision and therefore no need to rest simulation .But for a turtlebot env, i need to reset the simulation on collision? or is it not required and i stay with a similar implementation as yours with no resets?

rkojcev, 2018-01-23:
That might be true regarding the turtlebot, I would recommend to check it out experimentally and see, I have not used or tried ppo1 and turtlebot so I am not aware if resetting the simulation might be useful.